PSP-Archive/ARK-4 #401: Custom Launcher aborts

qwertzulan, 2024-09-15:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

Version 4.20.69 

### Homebrew/GAME (if applicable)

Custom Laucher

### Plugins ( if applicable )

cant install plugins 
i have an 512gb memory card filled with software and in the custom laucher if i scroll down fast after a few second it stucks...

have tried several attemps but everytime the same problem.

### Steps to reproduce

start psp
open custom launcher
press and hold down button (over 200 titles on it)

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-09-15:
512GB? Wasn't there a limit of 128GB?

In any case, the PSP itself is limited by RAM (32MB on 1K models, that's MB, not GB), so it won't be able to load that many games at once in a single list, you will have to categorize them in folders and use the File Browser as the launcher (there's a setting to set it as the main menu of the CL).

qwertzulan, 2024-09-15:
Look here how to format the microsdI have also tried a bigger one but it does not work.https://www.reddit.com/r/PSP/comments/13rmlf7/guide_format_a_sd_card_of_128gb_or_more/?tl=de#Can you tell me which setting I have to choose that this works.ThanksAm 15.09.24 um 14:58 schrieb JoseAaronLopezGarcia
                
            
            
                
                    Von: "JoseAaronLopezGarcia" ***@***.***>Datum: 15. September 2024An: "PSP-Archive/ARK-4" ***@***.***>Cc: "qwertzulan" ***@***.***>,"Author" ***@***.***>Betreff: Re: [PSP-Archive/ARK-4] Custom Launcher aborts (Issue #401)
                
                
512GB? Wasn't there a limit of 128GB?
In any case, the PSP itself is limited by RAM (32MB on 1K models, that's MB, not GB), so it won't be able to load that many games at once in a single list, you will have to categorize them in folders and use the File Browser as the launcher (there's a setting to set it as the main menu of the CL).

krazynez, 2024-09-15:
https://youtu.be/k76EmFxnYWo

https://docs.google.com/spreadsheets/d/1PJH87w-2Rn7Xk15KIuWYOKhmGYnk6tHr5BX3VOFBjWQ/edit?pli=1&gid=2115780760#gid=2115780760

Memory Cards have been thoroughly tested

qwertzulan, 2024-09-15:
It is formated correct.But in the custom launcher it hung up if scroll down because to much in there.How to categorize? Categories_lite.prx 1.7won’t work in xmb nor custom launcher Any ideas?Am 15.09.24 um 15:24 schrieb Brennen Murphy
                
            
            
                
                    Von: "Brennen Murphy" ***@***.***>Datum: 15. September 2024An: "PSP-Archive/ARK-4" ***@***.***>Cc: "Author" ***@***.***>,"qwertzulan" ***@***.***>Betreff: Re: [PSP-Archive/ARK-4] Custom Launcher aborts (Issue #401)
                
                
https://youtu.be/k76EmFxnYWo
https://docs.google.com/spreadsheets/d/1PJH87w-2Rn7Xk15KIuWYOKhmGYnk6tHr5BX3VOFBjWQ/edit?pli=1&gid=2115780760#gid=2115780760
Memory Cards have been thoroughly tested

krazynez, 2024-09-15:
Just do what Acid said and use File Brower (FB) in Custom Launcher (CL) to be the active window in the settings for CL. Then you can just launch the game you want from within the file browser.

qwertzulan, 2024-09-15:
That is exactly what I do there I have the problem Am 15.09.24 um 16:35 schrieb Brennen Murphy
                
            
            
                
                    Von: "Brennen Murphy" ***@***.***>Datum: 15. September 2024An: "PSP-Archive/ARK-4" ***@***.***>Cc: "qwertzulan" ***@***.***>,"Author" ***@***.***>Betreff: Re: [PSP-Archive/ARK-4] Custom Launcher aborts (Issue #401)
                
                
Just do what Acid said and use File Brower (FB) in Custom Launcher (CL) to be the active window in the settings for CL. Then you can just launch the game you want from within the file browser.

FULANO93, 2024-09-15:
I've already had a similar problem, but in my case what caused ARK to crash was an ISO that had somehow been corrupted. When I discovered what the ISO was, I copied it again and it resolved

qwertzulan, 2024-10-06:
I have tried several variations but always on the on the 45 iso it crashes. I have deleted some and tried again but always another one make the same problem. But always number 45! Maybe someone can help. ThanksAm 15.09.24 um 19:09 schrieb FULANO93
                
            
            
                
                    Von: "FULANO93" ***@***.***>Datum: 15. September 2024An: "PSP-Archive/ARK-4" ***@***.***>Cc: "qwertzulan" ***@***.***>,"Author" ***@***.***>Betreff: Re: [PSP-Archive/ARK-4] Custom Launcher aborts (Issue #401)
                
                
I've already had a similar problem, but in my case what caused ARK to crash was an ISO that had somehow been corrupted. When I discovered what the ISO was, I copied it again and it resolved

JoseAaronLopezGarcia, 2025-03-16:
PSP is known to have issues with memory cards above 128GB.

closed 2025-03-16 by JoseAaronLopezGarcia