PSP-Archive/ARK-4 #402: How to install classic cIPL?

Goosegit11, 2024-09-15:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20

### Steps to reproduce

Have 6.60 firmware
Launch cIPL installer

`Installing new cIPL
Only 6.61 fw is supported!`

End.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-09-15:
classic cIPL is only for 1K and 2K (minus 88v3)

Goosegit11, 2024-09-15:
Oh so that's what. I've heard that 6.60 is better in terms of compatibility with plugins or something like that. That is not the case now?

krazynez, 2024-09-15:
> Oh so that's what. I've heard that 6.60 is better in terms of compatibility with plugins or something like that. That is not the case now?

ARK-4 makes all plugins think its on 6.60 even if on 6.61

Goosegit11, 2024-09-15:
Great. I will update then, thanks!

closed 2024-09-15 by Goosegit11