PSP-Archive/ARK-4 #403: GTA Chinatown Wars wake time issues

Dooode, 2024-09-16:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.69 r15

### Homebrew/GAME (if applicable)

ULUS10490 GTA Chinatown Wars 

### Steps to reproduce

1. Launch GTA Chinatown Wars
2. Wait till it loads
3. Put PSP Go to sleep
4. Wake up
5. Watch the "checking media" loading screen for more than a minute

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

On

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

Dooode, 2024-09-16:
The same .iso file works properly on PS Vita Adrenaline and takes about a second to wake up.

krazynez, 2024-09-16:
It works, just have to wait a bit longer yeah. Wonder if there is a way to speed that up a bit though.

JoseAaronLopezGarcia, 2024-10-03:
Fixed.

Dooode, 2024-10-03:
Thanks a lot!