PSP-Archive/ARK-4 #405: Settings for ARK menu aren't saving after a restart.

gtaiscool236, 2024-09-26:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.69 r15

### Steps to reproduce

1. Change customisation settings in Settings from custom menu (NOT VSH).
2. Restart the PSP.

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-09-26:
I just tried this and works fine. Do you have ARK_01234 in savedata or are you relying on flash1 to save your settings?

gtaiscool236, 2024-09-26:
https://www.pspunk.com/psp-cfw/ I followed this guide to install ARK so it should be fine right? I double-checked and I did everything right.

krazynez, 2024-09-26:
Might be your memory stick. If you can try to use another one and recopy a fresh ARK_01234 folder into SAVEDATA.

JoseAaronLopezGarcia, 2024-09-26:
When in the launcher's settings menu you can press X to force save settings, though they should be saved automatically every time you exit the settings menu.

gtaiscool236, 2024-09-26:
> Might be your memory stick. If you can try to use another one and recopy a fresh ARK_01234 folder into SAVEDATA.

I am getting one delivered. Gonna take weeks tho lol

closed 2024-09-27 by gtaiscool236