PSP-Archive/ARK-4 #406: Sega Rally Revo - Infrastructure multiplayer doesnt work on Vita

Double-0-seven7, 2024-09-27:
### PSP/Vita Model

PCH-1000/1100 (Vita Fat), PCH-2000 (Vita Slim), VTE-1000 (Vita TV)

### ARK Version

4.20.69

### Homebrew/GAME (if applicable)

Sega Rally Revo

### Steps to reproduce

1.Set your DNS settings to 146.190.205.197
2.Connect to infrastructure mode
3.Play with someone else online
4.Game will disconnect on the Vita at the beginning of the race.
Note : May need the workaround/patch that got introduced in this issue #140 for the game to work.
Note 2: Game works fine on PSP with no issues.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-10-02:
Might be related to this issue: https://github.com/PSP-Archive/ARK-4/issues/32

Double-0-seven7, 2024-10-02:
> Might be related to this issue: #32

It's not a WEP/Open game it can use WPA to connect so it's not that.
That's why I think it might be related to CPU clocks again like the ATV issue I reported before.

JoseAaronLopezGarcia, 2025-03-18:
We're gonna move all further discussions on this matter to this new thread:
https://github.com/PSP-Archive/ARK-4/issues/519

closed 2025-03-18 by JoseAaronLopezGarcia

Double-0-seven7, 2025-03-18:
OK guess I will add it to that Issue as well.