PSP-Archive/ARK-4 #408: When using the plugin setup with GAMEID, the custom launcher freezes and the psp switches off

Xtremegamer13579, 2024-09-28:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.40.69 r15

### Homebrew/GAME (if applicable)

Gran Turismo

### Plugins ( if applicable )

temp AR

### Steps to reproduce

Pls try to stop the freezing and the cheatdevice for NFS Carbon OTC does not work when applying it
TempAR plugin not working on all games
I am not a person who knows about all the different codes of psp and scripting so i cannot tell you how to reproduce it

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-09-28:
We need to see the contents of your PLUGINS.TXT file (found inside the SEPLUGIN folder).

Xtremegamer13579, 2024-09-28:
[PLUGINS.TXT](https://github.com/user-attachments/files/17175795/PLUGINS.TXT)
Here it is

Xtremegamer13579, 2024-09-28:
Thanks for any help with my issue

Xtremegamer13579, 2024-09-28:
When i tried using pro c with my plugins it never work at all
I prefer ARK to Pro C ,It is much easier to use
I am also on discord so i see your messages
I will keep up with your updates
Thanks

krazynez, 2024-09-28:
> [PLUGINS.TXT](https://github.com/user-attachments/files/17175795/PLUGINS.TXT) Here it is

always, ms0:/SEPLUGINS/category_lite.prx, on 

is incorrect. [see](https://github.com/PSP-Archive/ARK-4/wiki/Recommended-Plugins#game-categories-lite)

Xtremegamer13579, 2024-09-28:
Category lite is working fine
It is the TempAr that isnt workong on any of my games

krazynez, 2024-09-28:
> Category lite is working fine It is the TempAr that isnt workong on any of my games

Yes because category lite is breaking it....

Xtremegamer13579, 2024-09-28:
So i must not connect the plugin to always
I must connect it to vsh only

maodus, 2024-09-28:
> So i must not connect the plugin to always I must connect it to vsh only

Yep, precisely.

Xtremegamer13579, 2024-09-28:
Trying it now

Xtremegamer13579, 2024-09-28:
Workin
Thank you so so much🫶🏻🫶🏻🫶🏻

closed 2024-09-28 by krazynez