PSP-Archive/ARK-4 #416: DCv10 NAND operations format utility not working

vrishe, 2024-10-15:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.69 r150

### Steps to reproduce

Greetings!

1. Make a magic stick with MagicMemoryCreator
2. Boot it with pysweeper
3. Proceed towards NAND operations -> Format Lflash
4. Check the volume sizes, navigate right, confirm

Doing so one will get a message 'Error loading updater modules.' overlapping an empty progressbar.
But sometimes there is an empty messagebox, without any text, just a circle glyph (all that's left of 'Back' hint) at the bottom.

Tried this with several SDcards and a couple MS Duo adapters in different combinations. The result stays the same.

PSP-1004 TA-082 [g01]

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-10-15:
Fixed: https://github.com/PSP-Archive/ARK-4/releases/download/v1729022821/ARK-4-v1729022821.zip

vrishe, 2024-10-15:
It works now. A big thanks! :)

closed 2024-10-15 by vrishe