PSP-Archive/ARK-4 #417: Support for the Resistance Remastered Controls plugin by TheOfficialFloW

AdrCant, 2024-10-20:
### New Feature

This is the first time I’m opening an issue thread; I wasn't sure where to put it, and the bugs section doesn’t allow me to explain what’s happening, so I decided to post it here. Thank you in advance.

I’ve encountered a problem with this collection of plugins designed for those who own a PS Vita with Adrenaline. Its purpose is to enable full support for the second analog stick of the Vita in the following games:

— Grand Theft Auto: Liberty City Stories
— Grand Theft Auto: Vice City Stories
— Lara Croft: Tomb Raider Legend
— Lara Croft: Tomb Raider Anniversary
— Assassin's Creed: Bloodlines (This was removed by the author; I assume it was because it interfered with one of the tutorial sections, as I don’t see any other reason for its removal)
— Tom Clancy's Splinter Cell: Essentials
— Resistance: Retribution
— Prince of Persia: Revelation
— Prince of Persia: Rival Swords
— Kingdom Hearts Birth By Sleep (+ Final Mix)
— Metal Gear Solid Peace Walker
— Metal Gear Solid Portable Ops
— The Warriors
— Tony Hawk Project 8

I’ve been testing them, and from what I’ve seen, all of them work well except for the one for Resistance Retribution; this one simply doesn’t work at all. I’ve tried installing it both in the GAME section and by GAME ID, but it didn’t help at all. This plugin makes the most changes by taking advantage of a feature that the game already has. What it does is trick the game into thinking you’re playing with a PS3 controller, activating Plus mode by assigning R2 and L2 to the right and left buttons of the D-pad, and assigning R3 to down. It also provides a way to unlock the Infected mode, which is normally unlocked by connecting the game to a copy of Resistance 2, simply by placing an empty file titled resistance_infected.bin in seplugins.

Would it be possible to make this plugin work correctly in ARK-4?

Links of interest:

TheOfficialFloW’s GitHub page about this collection of plugins: https://github.com/TheOfficialFloW/RemasteredControls

VK page of the PG Community, which compiled all the plugins, including the one for Assassin's Creed (in Russian): https://vk.com/@portablegaming-remastered

JoseAaronLopezGarcia, 2024-10-20:
The plugin works fine, I used it myself and haven't had an issue, is arkrightanalog (native vita plugin) installed?

AdrCant, 2024-10-20:
No, I didn't know about that plugin. I'm going to install it and check if it solves the problem.

JoseAaronLopezGarcia, 2024-10-20:
It should have been already installed by FasterARK, unless you did a manual installation.

AdrCant, 2024-10-20:
I installed it via Adrenaline; I didn't use FasterArk.

krazynez, 2024-10-20:
> I installed it via Adrenaline; I didn't use FasterArk.

Did you correctly install the plugin via ARK-4 then. We use a diffrent plugin system than adrenaline does. It does look like aftwards there are some extra steps that the fl0w states in his readme also.

AdrCant, 2024-10-20:
I installed it following the method indicated in the corresponding section of the wiki: https://github.com/PSP-Archive/ARK-4/wiki/Plugins

AdrCant, 2024-10-20:
> > I installed it via Adrenaline; I didn't use FasterArk.
> 
> Did you correctly install the plugin via ARK-4 then. We use a diffrent plugin system than adrenaline does. It does look like aftwards there are some extra steps that the fl0w states in his readme also.

The collection has not been updated since 2020; no new instructions have been added.

krazynez, 2024-10-20:
> The collection has not been updated since 2020; no new instructions have been added.

I mean to make sure nothing is mapped to right analog as it states in his readme. But guess Acid need to test on ADR to see if it works there.

AdrCant, 2024-10-20:
I tried installing via FasterArk, in case it was an issue exclusive to the Adrenaline installation, but it didn’t work either.

AdrCant, 2024-10-21:
> I mean to make sure nothing is mapped to right analog as it states in his readme. But guess Acid need to test on ADR to see if it works there.

I followed all the steps described by thefl0w, but it still doesn't work. It seems strange that it's the only plugin in the entire collection that's having issues.

JoseAaronLopezGarcia, 2024-10-21:
> I followed all the steps described by thefl0w, but it still doesn't work. It seems strange that it's the only plugin in the entire collection that's having issues.

Please note that plugins on ARK are installed a bit differently than on Adrenaline, so game.txt won't work.

AdrCant, 2024-10-21:
> Please note that plugins on ARK are installed a bit differently than on Adrenaline, so game.txt won't work.

I mean that I followed the given instructions to avoid any conflict (to check that there was nothing mapped for the right stick). I installed the plugin by following the steps indicated in the wiki.

JoseAaronLopezGarcia, 2024-10-22:
> I mean that I followed the given instructions to avoid any conflict (to check that there was nothing mapped for the right stick). I installed the plugin by following the steps indicated in the wiki.

Ok I'll check it out, maybe there is an issue. I know that I use the gta_remastered plugin and the camera_patch_lite plugin just fine with ARK, but maybe the resistance one isn't playing nice.

AdrCant, 2024-11-07:
Is there any update? Was the cause of the problem found?

AdrCant, 2025-02-14:
It’s been a while, has there been any update regarding the issue? I’ve tried again with the latest version of ARK-4, and the problem is still there, the plugin still isn’t working. If it helps, I performed the test with the US version of Resistance Retribution (ID UCUS-98668).