PSP-Archive/ARK-4 #421: PRO Online client loaded but not working

PiporGames, 2024-11-07:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r150

### Homebrew/GAME (if applicable)

Might happen with any Adhoc Game, tried with UCES-01312 and NPJH50263

### Plugins ( if applicable )

game, ms0:/seplugins/atpro.prx, on
game, ms0:/seplugins/AdHocRedirectorWiFi.prx, off
game, ms0:/seplugins/AdHocToUSB.prx, off
game, ms0:/seplugins/dayviewer_game.prx, on
game, ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx, off
vsh, ms0:/seplugins/dayviewer.prx, off
vsh, ms0:/seplugins/mp3play_lite.prx, off
vsh, ms0:/seplugins/XIVSephirothXIV_Patcher.prx, on
always, ms0:/seplugins/RemoteJoyLite.prx, off
always, ms0:/seplugins/660_plugins_on_661.prx, on
xmb, ms0:/seplugins/category_lite.prx, off

### Steps to reproduce

1. Enable atpro plugin
2. Load into game
3. Select Ad-Hoc Option
4. Either the game reports a connection error has ocurred, or error 80010002 is shown
5. Status on the PS menu says "Waiting for ADHOC SSID setup..."

Note: PRO Online worked on this same device on PRO-C 6.61, with the same games.

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-11-14:
PRO Online is pretty finicky, it's hard to get fully working and depends on many things, including your wifi hotspot.
There's some guides online that might help you out.

PiporGames, 2024-11-14:
I know. I have tried many and, as I've said, it works ON PRO-C. It won't work under the same exact scenario on ARK-4.
And, since ARK-4 features "PRO Online Client" as one of it's features that's why I'm submitting a bug report.

JoseAaronLopezGarcia, 2024-11-14:
Can you elaborate further on the test scenario you've done?

PiporGames, 2024-11-20:
Sorry, my PSP's power on switch got stuck and boot loops the console. I can't reproduce or give exact details right now, but I will try to provide some context.

In PRO-C some games would work and others would crash, maybe due to insufficient memory. The thing is that the game would recognise the AdHoc option and would at least try to setup a connection/lobby.

In ARK-4, the game instantly declines the network setup petition altogether. Games that notify the user about this usually handles the error as a generic connection error. Others like Ace Combat spawn a msgbox with said error.

I tried disabling all plugins, with and without the 6.60 on 6.61 plugins support, and with the recommend settings. I've also tried both with a real WiFi router and mobile data streamed from my phone. Both don't work, as the plugin doesn't even get to start the network setup.

This is also backed up by the fact that the PS Menu states that the plugin is waiting for an AdHoc setup connection... Which won't ever start because the game won't allow said connection in the first place.

I'm thinking maybe the entry point for the adhoc patch on the plugin is invalid but I don't really know.
I might install PRO-C again and see if the problem still persists. I'm not using a permanent CFW.
I can't give specific hardware details until I fix the console.

Hope it helps :)

PiporGames, 2024-12-03:
It's already activated, as seen on the bug report.
Btw I just fixed my console just hours ago. I will be testing with different settings and see what comes through.

PiporGames, 2024-12-10:
I've tested with a clean MS and a clean copy of both ARK and PRO-C and... weird.
Now it seems like PRO-C is the one not working correctly and ARK is.
ARK now does setup the network and connects to socom.cc.
PRO-C won't do that UNLESS the game is booted with the WLAN switch off, disabling the PRO Chat.
It might seem that the plugin is disabled if the game is booted with the WLAN switch on off, as stated on the text in the PS Menu, but atlast, the game will try and succeed in connecting to the server even if the PRO Chat is disabled.

Seems like I will need to keep investigating this.
For now, the issue is resolved, as a clean copy of ARK makes the PRO plugin work.

closed 2024-12-10 by PiporGames