PSP-Archive/ARK-4 #422: Custom launcher freezes on 'view info' for some games

dopeyhex, 2024-11-14:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r150

### Homebrew/GAME (if applicable)

Daedalus x64

### Steps to reproduce

1. Open the menu in the games tab of the custom launcher with L button for any emulator or program like FuSa (anything in the second column really)
2. Select "view info"
3. The launcher will either freeze or show the same game ID across all of the column

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-12-17:
It's not really frozen, it's just calculating the size of all files, and some homebrew have a lot of those, so it can be slow.
This can't really be fixed, either we get rid of the recursive size calculation (and just report the size of the eboot only, though that would be incomplete data) or we add some animation to make it clear that it's working on it.

krazynez, 2024-12-17:
Can you try this on the latest pre-release and see if you still have the issue.