PSP-Archive/ARK-4 #423: File browser in custom launcher hangs up the system if any folder from PSP/GAME is chosen

dopeyhex, 2024-11-14:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r150

### Steps to reproduce

1. Open the Files tab
2. Go to PSP/GAME
3. Select any folder below the .//, in my case it's ARK_CIPL/
4. System turns off, seems to be a memory leak

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-11-14:
Yes this is know, I have a slight temp fix but not fully so never pushed the changes to test yet. About to start working on it again.

krazynez, 2024-11-14:
Fixed: https://github.com/PSP-Archive/ARK-4/releases/download/v1731624201/ARK-4-v1731624201.zip

Though not sure what else we fixed so might just want to copy VBOOT.PBP from the ARK_01234 folder within in to your ARK_01234 on your memory stick.

dopeyhex, 2024-11-15:
Worked like a charm, thanks a million!

closed 2024-11-15 by dopeyhex