PSP-Archive/ARK-4 #425: Error when trying to open UMD Videos

dantas-arthur, 2024-11-15:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

v1731624201

### Steps to reproduce

1. Place any UMD Video in the ISO folder
2. Press SELECT, in the Advance Menu select the option ISO Video Mount (does not locate the video)
3. Open ARK-4 Custom Launcher
4. Try to open the UMD Video
5. The console stays on a black screen and returns to the XMB

https://github.com/user-attachments/assets/0535afab-ddef-4307-8a55-1938f2a037bf

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-11-15:
You can't run UMD VIdeos with Custom Launcher. It being in a game runlevel it will not detect the proper files to actually reboot and load the video.

dantas-arthur, 2024-11-15:
> You can't run UMD VIdeos with Custom Launcher. It being in a game runlevel it will not detect the proper files to actually reboot and load the video. 

I also tried playing the video through the XMB, opened the Advanced Menu, and tried mounting the file, but it only shows 'None' What's strange is that in the Custom Launcher, the video displays normally; I just can't open it. The problem isn't with the file I'm using either, because I used the same file normally on Infinity.

dantas-arthur, 2024-11-16:
> You can't run UMD VIdeos with Custom Launcher. It being in a game runlevel it will not detect the proper files to actually reboot and load the video.

Oh, I believe I found the issue. I placed the .iso video file directly in the ISO folder of the memory card, and I think that since I was using the Category Lite Plugin, it wasn’t allowing the video to be located in the Advanced Menu. I created a 'video' folder inside the ISO folder, added the file there, and now it’s working fine! Thank you for your help, Krazynez, you can close this issue now.

closed 2024-11-16 by krazynez