PSP-Archive/ARK-4 #426: PSP-E1004 | Can’t install Ark-4 / Can‘t Open VSH

averygoodtime, 2024-11-16:
### PSP/Vita Model

PSP-E1000 (Street)

### ARK Version

Version 4.20.69

### Steps to reproduce

1. Download latest release 
2. Connect to pc 
3. Copy ark loader, ark 01234, ark cipl / ark Full installer
4. Safely Remove 
5. Start ark4 live 
6. Start cipl 
7. Press install
8. be in live 
9. Basically Same with Full installer except Reboots in Stock fw (6.61, tried 6.60 aswell)

getting no error Message or anything. Tried with original Memory Card as well as sd Adapter.

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

On

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

averygoodtime, 2024-11-16:
Sorry forgot to mention while making sure it doesn't work now for no reason:
VHS Menu only opens on force start After soft reboot and is unresponsive. Closes again and makes system responsive again After another soft reboot.

krazynez, 2024-11-17:
Just sounds like a button is stuck

averygoodtime, 2024-11-17:
> Just sounds like a button is stuck

Any Button in particular? In Games as well as in the Menu all Buttons seem to work Like normal. 

However I might disassemble and clean it again.

Any other ideas as well?

krazynez, 2024-11-17:
You can try a button tester out.  https://www.gamebrew.org/wiki/KeysTester_PSP

closed 2024-11-17 by averygoodtime