PSP-Archive/ARK-4 #430: Custom Launcher not loading games when cat lite is disabled,wwe smackdown vs raw 2011,when starting the game,there is black screen and the psp switches off

Karim0786, 2024-12-06:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r15

### Homebrew/GAME (if applicable)

WWE Smackdown Vs Raw

### Plugins ( if applicable )

[PLUGINS.TXT](https://github.com/user-attachments/files/18041631/PLUGINS.TXT)

### Steps to reproduce

I dont know how

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

Karim0786, 2024-12-07:
Thank you working now,just sad about the game

JoseAaronLopezGarcia, 2024-12-07:
> Thank you working now,just sad about the game

You can install the latest pre-release to fix the game.

JoseAaronLopezGarcia, 2024-12-17:
this has been fixed and is available in latest release

closed 2024-12-17 by JoseAaronLopezGarcia