PSP-Archive/ARK-4 #431: Wrong translation on Chinese VSH Menu

Tsubamei, 2024-12-07:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

ARK 4.20.69 R150

### Steps to reproduce

[image]

The text in the fourth line should be in the third line.

The text in the fourth line **UMD 区域** from the bottom should be **导入旧版插件** .
The text in the third line **禁用休眠** from the bottom should be **重置ARK设置** .

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-12-07:
Can you do us a favor. We both can't read Chinese so can you redorder this :

 https://github.com/PSP-Archive/ARK-4/blob/a9ac6a12e581a97ffd14916d744ab1e22f9e545b/extras/menus/arkMenu/themes/translations/resources/satelite_chs_utf8.txt

to align with the english version:

https://github.com/PSP-Archive/ARK-4/blob/a9ac6a12e581a97ffd14916d744ab1e22f9e545b/extras/menus/vshmenu/trans_en.c#L24

We would appreatate that very much. Just had to use google translate for most so alignment probably got jacked up. If there is any others that you can do by all means we will be happy to add as well.

krazynez, 2024-12-14:
Here can you verify this is good for us? : https://github.com/PSP-Archive/ARK-4/releases/download/v1734141236/ARK-4-v1734141236.zip

Tsubamei, 2024-12-15:
I'd be happy to assist. I proofread both the simplified and traditional Chinese translations, and presented the correct terms in a table for easy reference.
(A problem is that I can't find where the word "Toggle" appears. I only have a PSP-3000 and use the R151 version.)

| English                 | Simplified Chinese | Traditional Chinese |
| ----------------------- | ------------------ | ------------------- |
| Default,                | 默认                 | 預設                  |
| Japan,                  | 日本                 | 日本                  |
| America,                | 美国                 | 美國                  |
| Europe,                 | 欧洲                 | 歐洲                  |
| Korea,                  | 韩国                 | 南韓                  |
| United Kingdom,         | 英国                 | 英國                  |
| Latin America,          | 拉丁美洲               | 拉丁美洲                |
| Australia,              | 澳大利亚               | 澳洲                  |
| Hong Kong,              | 香港                 | 香港                  |
| Taiwan,                 | 台湾                 | 臺灣                  |
| Russia,                 | 俄罗斯                | 俄羅斯                 |
| China,                  | 中国                 | 中國                  |
| Debug I,                | Debug I            | Debug I             |
| Debug II,               | Debug II           | Debug II            |
| Disabled,               | 已禁用                | 關閉                  |
| Enabled,                | 已启用                | 開啟                  |
| Normal,                 | 普通                 | 普通                  |
| Toggle,                 | 切换                 | 切換                  |
| X -> O,                 | X确认/O返回            | X決定/O返回             |
| O -> X,                 | O确认/X返回            | O決定/X返回             |
| USB DEVICE,             | USB 设备             | USB 裝置              |
| USB READONLY,           | USB 只读             | USB 唯讀              |
| ISO VIDEO MOUNT,        | 挂载 ISO 影片          | 掛接 ISO 影片           |
| FG COLORS,              | 字体颜色               | 字型色彩                |
| BG COLORS,              | 背景颜色               | 背景色彩                |
| FONT,                   | 字体                 | 字型樣式                |
| MENU DESIGN,            | 菜单样式               | 選單設計                |
| MAIN MENU,              | 主菜单                | 主選單                 |
| CONVERT BATTERY,        | 电池转换               | 電池轉換                |
| SWAP X/O BTNS,          | 交换 X/O 按键          | 交換 X/O 按鈕           |
| VSH REGION,             | VSH 区域             | VSH 區域              |
| UMD REGION,             | UMD 区域             | UMD 區域              |
| IMPORT CLASSIC PLUGINS, | 导入旧版插件             | 注入舊版插件              |
| ACTIVATE FLASH AND WMA, | 激活 FLASH 和 WMA     | 啟動 Flash 和 WMA      |
| DELETE HIBERNATION,     | 禁用休眠               | 禁用休眠                |
| RESET ARK SETTINGS,     | 恢复 ARK 默认设置        | 恢復 ARK 預設           |
| BOOT RANDOM ISO,        | 启动随机 ISO           | 運行隨機 ISO            |
| GO BACK,                | 返回                 | 返回                  |
| NO HIBERNATION SUPPORT, | 不支持休眠              | 不支援休眠               |
| Normal -> Pandora,      | 普通电池 -> 神奇电池       | 普通電池 -> 潘朵拉電池       |
| Pandora -> Normal,      | 神奇电池 -> 普通电池       | 潘朵拉電池 -> 普通電池       |
| VSH MENU,               | VSH 菜单             | VSH 選單              |
| ADVANCED VSH MENU,      | 高级 VSH 菜单          | 進階 VSH 選單           |
| RECOVERY MENU ->,       | 恢复菜单 ->            | 還原選單 ->             |
| SHUTDOWN DEVICE,        | 关机                 | 關機                  |
| SUSPEND DEVICE,         | 睡眠模式               | 睡眠模式                |
| RESET DEVICE,           | 重启主机               | 重新啟動                |
| RESET VSH,              | 重启 VSH             | 重啟 VSH              |
| EXIT,                   | 退出                 | 退出                  |
| Unsupported,            | 不支持                | 不支援                 |
| Adrenaline USB,         | 肾上腺素 USB           | 腎上腺素 USB            |
| Flash,                  | Flash              | Flash               |
| UMD,                    | UMD                | UMD                 |
| Internal Storage,       | 内部存储               | 內部記憶體               |
| Memory Stick,           | 记忆棒                | MS 記憶卡              |
| None,                   | 无                  | 無                   |
| Classic,                | 经典                 | 經典                  |
| Simple,                 | 简洁                 | 簡單                  |
| Advanced,               | 高级                 | 高級                  |
| New,                    | 新版                 | 新式                  |
| Random,                 | 随机                 | 隨機                  |
| Red,                    | 红色                 | 紅色                  |
| Light Red,              | 浅红色                | 淡紅色                 |
| Orange,                 | 橙色                 | 橙色                  |
| Light Orange,           | 浅橙色                | 淺橙色                 |
| Yellow,                 | 黄色                 | 黃色                  |
| Light Yellow,           | 浅黄色                | 淺黃色                 |
| Green,                  | 绿色                 | 綠色                  |
| Light Green,            | 浅绿色                | 淺綠色                 |
| Blue,                   | 蓝色                 | 藍色的                 |
| Light Blue,             | 浅蓝色                | 淺藍色                 |
[... cut ...]
| Aqua,                   | 水绿色                | 水藍色                 |
| Light Aqua,             | 浅水绿色               | 淺水藍色                |
| Grey,                   | 灰色                 | 灰色                  |
| Light Grey,             | 浅灰色                | 淺灰色                 |
| Black,                  | 黑色                 | 黑色                  |
| Light Black,            | 浅黑色                | 淺黑色                 |
| White,                  | 白色                 | 白色                  |
| Light White             | 浅白色                | 淺白色                 |

krazynez, 2024-12-15:
Omg that is perfect thank you so much I will update them both! Super awesome of you!

krazynez, 2024-12-15:
> I'd be happy to assist. I proofread both the simplified and traditional Chinese translations, and presented the correct terms in a table for easy reference. (A problem is that I can't find where the word "Toggle" appears. I only have a PSP-3000 and use the R151 version.)
> English 	Simplified Chinese 	Traditional Chinese
> Default, 	默认 	預設
> Japan, 	日本 	日本
> America, 	美国 	美國
> Europe, 	欧洲 	歐洲
> Korea, 	韩国 	南韓
> United Kingdom, 	英国 	英國
> Latin America, 	拉丁美洲 	拉丁美洲
> Australia, 	澳大利亚 	澳洲
> Hong Kong, 	香港 	香港
> Taiwan, 	台湾 	臺灣
> Russia, 	俄罗斯 	俄羅斯
> China, 	中国 	中國
> Debug I, 	Debug I 	Debug I
> Debug II, 	Debug II 	Debug II
> Disabled, 	已禁用 	關閉
> Enabled, 	已启用 	開啟
> Normal, 	普通 	普通
> Toggle, 	切换 	切換
> X -> O, 	X确认/O返回 	X決定/O返回
> O -> X, 	O确认/X返回 	O決定/X返回
> USB DEVICE, 	USB 设备 	USB 裝置
> USB READONLY, 	USB 只读 	USB 唯讀
> ISO VIDEO MOUNT, 	挂载 ISO 影片 	掛接 ISO 影片
> FG COLORS, 	字体颜色 	字型色彩
> BG COLORS, 	背景颜色 	背景色彩
> FONT, 	字体 	字型樣式
> MENU DESIGN, 	菜单样式 	選單設計
> MAIN MENU, 	主菜单 	主選單
> CONVERT BATTERY, 	电池转换 	電池轉換
> SWAP X/O BTNS, 	交换 X/O 按键 	交換 X/O 按鈕
> VSH REGION, 	VSH 区域 	VSH 區域
> UMD REGION, 	UMD 区域 	UMD 區域
> IMPORT CLASSIC PLUGINS, 	导入旧版插件 	注入舊版插件
> ACTIVATE FLASH AND WMA, 	激活 FLASH 和 WMA 	啟動 Flash 和 WMA
> DELETE HIBERNATION, 	禁用休眠 	禁用休眠
> RESET ARK SETTINGS, 	恢复 ARK 默认设置 	恢復 ARK 預設
> BOOT RANDOM ISO, 	启动随机 ISO 	運行隨機 ISO
> GO BACK, 	返回 	返回
> NO HIBERNATION SUPPORT, 	不支持休眠 	不支援休眠
> Normal -> Pandora, 	普通电池 -> 神奇电池 	普通電池 -> 潘朵拉電池
> Pandora -> Normal, 	神奇电池 -> 普通电池 	潘朵拉電池 -> 普通電池
> VSH MENU, 	VSH 菜单 	VSH 選單
> ADVANCED VSH MENU, 	高级 VSH 菜单 	進階 VSH 選單
> RECOVERY MENU ->, 	恢复菜单 -> 	還原選單 ->
> SHUTDOWN DEVICE, 	关机 	關機
> SUSPEND DEVICE, 	睡眠模式 	睡眠模式
> RESET DEVICE, 	重启主机 	重新啟動
> RESET VSH, 	重启 VSH 	重啟 VSH
> EXIT, 	退出 	退出
> Unsupported, 	不支持 	不支援
> Adrenaline USB, 	肾上腺素 USB 	腎上腺素 USB
> Flash, 	Flash 	Flash
> UMD, 	UMD 	UMD
> Internal Storage, 	内部存储 	內部記憶體
> Memory Stick, 	记忆棒 	MS 記憶卡
> None, 	无 	無
> Classic, 	经典 	經典
> Simple, 	简洁 	簡單
> Advanced, 	高级 	高級
> New, 	新版 	新式
> Random, 	随机 	隨機
> Red, 	红色 	紅色
> Light Red, 	浅红色 	淡紅色
> Orange, 	橙色 	橙色
> Light Orange, 	浅橙色 	淺橙色
> Yellow, 	黄色 	黃色
> Light Yellow, 	浅黄色 	淺黃色
> Green, 	绿色 	綠色
> Light Green, 	浅绿色 	淺綠色
> Blue, 	蓝色 	藍色的
> Light Blue, 	浅蓝色 	淺藍色
> Indigo, 	靛蓝色 	靛青色
> Light Indigo, 	浅靛蓝色 	淺靛青色
> Violet, 	紫罗兰色 	紫羅蘭色
> Light Violet, 	浅紫罗蓝色 	淺紫羅蘭色
> Pink, 	粉色 	粉紅色
> Light Pink, 	浅粉色 	淺粉紅色
> Purple, 	紫色 	紫色
> Light Purple, 	浅紫色 	淺紫色
> Teal, 	蓝绿色 	水鴨色
> Light Teal, 	浅蓝绿色 	淺水鴨色
> Aqua, 	水绿色 	水藍色
> Light Aqua, 	浅水绿色 	淺水藍色
> Grey, 	灰色 	灰色
> Light Grey, 	浅灰色 	淺灰色
> Black, 	黑色 	黑色
> Light Black, 	浅黑色 	淺黑色
> White, 	白色 	白色
> Light White 	浅白色 	淺白色

Pretty sure Toggle is just from the original avshmenu where we used it to for other options that got replaced with other ways. Like X -> O used to say Toggle X- > O

krazynez, 2024-12-17:
Should be fixed in latest pre-release now. Thanks again for the full list.

closed 2024-12-17 by krazynez

Tsubamei, 2024-12-19:
> Should be fixed in latest pre-release now. Thanks again for the full list.

Thanks for the update. I've updated to the latest version and checked, and there are no more wrongs.
And I found some minor errors in the Japanese translation, I'm attaching the corrections below.

| English       | Japanese |
| ------------- | -------- |
| Indigo,       | インディゴ    |
| Light Indigo, | ライトインディゴ |
| Violet,       | スミレ      |
| Light Violet, | ライトスミレ   |
| Pink,         | ピンク      |
| Light Pink,   | ライトピンク   |
| Purple,       | パープル     |
| Light Purple, | ライトパープル  |
| Teal,         | ティール     |
| Light Teal,   | ライトティール  |
| Aqua,         | アクア      |
| Light Aqua,   | ライトアクラ   |

krazynez, 2024-12-21:
> インディゴ
> Light Indigo, 	ライトインディゴ
> Violet, 	スミレ
> Light Violet, 	ライトスミレ
> Pink, 	ピンク
> Light Pink, 	ライトピンク
> Purple, 	パープル
> Light Purple, 	ライトパープル
> Teal, 	ティール
> Light Teal, 	ライトティール
> Aqua, 	アクア
> Light Aqua, 	ライトアクラ

Nice thanks updated.