PSP-Archive/ARK-4 #437: ARK Launcher: Add mute indicator

Aneemasta, 2024-12-18:
### New Feature

Apart from 31 levels of audio volume (zero included), there is also a mute mode activated by `Note` long-press. Could you please add a mute indicator similar to the one from the XMB?
1. The original indicator:
[image]
2. How it may look in the `ARK Launcher`:
[image]
3. The original icon (extracted from screenshot):
[image]

krazynez, 2024-12-18:
Yeah this is a good idea.

Aneemasta, 2024-12-20:
> The only issue is ALL themes would need updates however
Probably, there could be a fallback to show nothing instead of MUTE.PNG icon when there's no one within the theme to keep compatibility with old themes.

krazynez, 2024-12-21:
It is now in the latest pre-release. Acid knocked me against the head on that one forgot about the ya2d lib

closed 2024-12-21 by krazynez

Aneemasta, 2024-12-21:
I see. So, you're taking a `MUSIC.PNG` from the theme with the red cross before it. Pretty good, I must say!

Thanks!