PSP-Archive/ARK-4 #438: PSP is crashes when MS IPL not found by pressing and hold the L trigger.

lala1818, 2024-12-20:
### PSP/Vita Model

Any

### ARK Version

Latest

### Steps to reproduce

4.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-12-20:
Boots into XMB as expected for me.

JoseAaronLopezGarcia, 2025-03-16:
Closing as it's not possible to reproduce.

closed 2025-03-16 by JoseAaronLopezGarcia