PSP-Archive/ARK-4 #439: Add feature when press and hold L trigger + Home button to launch DC MS IPL

lala1818, 2024-12-22:
### New Feature

L trigger is included of the key combo of 11g Street model when enter service mode. But Home is not. That's why need to add Home button for L trigger.

krazynez, 2024-12-22:
[image]

JoseAaronLopezGarcia, 2024-12-22:
I don't understand the rationale either. Why is L+Home preferable to only L?

lala1818, 2024-12-22:
Because it is crashes when MS IPL is not found. If you try to enter service mode on 11g then it is will crashes and can't unbrick. The key combo is needs to hold the L trigger. I know because I has tryed on 11g model.

JoseAaronLopezGarcia, 2024-12-22:
None of that makes any sense. cIPL won't crash simply because the file isn't there, that's just not true, and even if the file is not there then that means DC10 is not installed, in which case, what are you even trying to enter service mode for? What could you possibly achieve by entering service mode without DC10 anyways?

krazynez, 2024-12-22:
> None of that makes any sense. cIPL won't crash simply because the file isn't there, that's just not true, and even if the file is not there then that means DC10 is not installed, in which case, what [...]

Agreed I have personally tested this the other day and it just boot into XMB as expected.

lala1818, 2024-12-22:
https://files.fm/u/y4uwpmuucp

JoseAaronLopezGarcia, 2024-12-22:
Just install DC... I keep failing to see what the issue is.

lala1818, 2024-12-22:
Im edited because the home button is to boot into OFW.

krazynez, 2024-12-22:
This issue still makes sense.

MizeryUwU, 2024-12-31:
Sorry guys maybe you just didn't understand him considering English is not his native language. Anyway, is holding the left trigger button on boot supposed to direct you to the ARK-DC recovery menu even without baryon sweeper/Pandora battery? For context, I hacked my gow special edition, psp 2k with the latest version of ark-4 and CIPL and installed Despertar Del Cementerio as well but jut like this guy, couldn't boot the CD recovery menu while holding the left trigger. What are we doing wrong? Or is the recovery menu designed not to boot at all except when we actually brick our psp and unbricik it by using a baryon sweeper?

krazynez, 2024-12-31:
> Sorry guys maybe you just didn't understand him considering English is not his native language. Anyway, is holding the left trigger button on boot supposed to direct you to the ARK-DC recovery menu even without baryon sweeper/Pandora battery? For context, I hacked my gow special edition, psp 2k with the latest version of ark-4 and CIPL and installed Despertar Del Cementerio as well but jut like this guy, couldn't boot the DC recovery menu while holding the left trigger. What are we doing wrong? Or is the recovery menu designed not to boot at all except when we actually brick our psp and decide to unbrick it by using a baryon sweeper?

Yes holding L trigger after DC ARK as been install and using the new cIPL should boot into DCARK. If it is not can you please run [Yoti's PSPIdent](https://github.com/Yoti/psp_pspident/releases/tag/v3.1) and upload the screenshot.

MizeryUwU, 2024-12-31:
[image]
Here

lala1818, 2024-12-31:
My all PSP 02g, 04g, 09g and 11g is freezes when I holding L Trigger and memory stick not inserted or no DC10 installed on memory stick. :( And if you forgot to put out the battery then it is will kill the battery's PCB. On Street is lucky because it is just freezes the syscon and need to remove the 11g battery to reset syscon, then put back and it works fine ;) So guys it is needs to keep in mind and understand the L trigger to load MS IPL is not safe for PSP and battery (mostly battery). I hope Acid is will fix that issue soon as possible. There is not load XMB when IPL not found on MS or no MS inserted. I'm really like what doing the developers. It is a nice thing. I fully respect you guys. :) Me too don't understand why not boot XMB when MS IPL not found.

MizeryUwU, 2024-12-31:
What do you suggests should I do? Should I reinstall DC? I am pretty sure it was installed properly though. I got new CIPL too

lala1818, 2024-12-31:
MizeryUwU 
1. Format your Memory Stick with PSP Tool. 
2. Copy the ARK-DC to PSP/GAME folder then run via XMB.

krazynez, 2024-12-31:
Okay I see the issue, one moment, its missing the checks. I am going to have to go add all models again it seems like for msipl. But I will add this one and you can test it about to push to pre-release. I will supply the link to the zip once it is done compiling in Github Actions and then just rerun DCARK Installer again with the new one in the zip file.

MizeryUwU, 2024-12-31:
Thank you kind sir! @krazynez

krazynez, 2024-12-31:
https://github.com/PSP-Archive/ARK-4/releases/download/v1735653724/ARK-4-v1735653724.zip

Rerun DCARK with the one supplied there it should work afterwards. Let me know if it does/doesn't.

krazynez, 2024-12-31:
> I shouldn't do any prerequisites right? Like uninstalling the previous DC?

You can just run it over the other one it will rewrite the files.

MizeryUwU, 2024-12-31:
Sadly, after rerunning the DC installer it still doesn't work 😢

lala1818, 2024-12-31:
[image]
[image]
[image]
[image]

krazynez, 2024-12-31:
> Sadly, after rerunning the DC installer it still doesn't work 😢

What exactly are you doing? Can you provided me a step by step. I also did some cleanup so try the [latest pre-realease](https://github.com/PSP-Archive/ARK-4/releases/download/v1735656052/ARK-4-v1735656052.zip) again. I just tested on my 05g ( GO ) with no issues. Perhaps you should delete the TM folder at the root of your memory stick first.

krazynez, 2024-12-31:
> ![02g PSP Ident](https://private-user-images.githubusercontent.com/192350063/399484624-4312c022-e1ea-47ad-82f3-b7ce34c40a67.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2NTYzOTcsIm5iZiI6MTczNTY1NjA5NywicGF0aCI6Ii8xOTIzNTAwNjMvMzk5NDg0NjI0LTQzMTJjMDIyLWUxZWEtNDdhZC04MmYzLWI3Y2UzNGM0MGE2Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMVQxNDQxMzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYzBmNWVmMjdlN2JmYWZkZmEwYWVkODRiNTE1YTc3MmJhYzY4MmE2ODc4M2IyODU5ZTFjYmM0YjAzZDA5NDFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.H42KHbTTutMRTR6Fz4hxv28dLunS5nRr2333IXIVpxE) ![09g PSP Ident](https://private-user-images.githubusercontent.com/192350063/399484667-d583b986-9d58-4f96-9d9b-9009b0fb9633.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2NTYzOTcsIm5iZiI6MTczNTY1NjA5NywicGF0aCI6Ii8xOTIzNTAwNjMvMzk5NDg0NjY3LWQ1ODNiOTg2LTlkNTgtNGY5Ni05ZDliLTkwMDliMGZiOTYzMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMVQxNDQxMzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZTczY2E4NWM5YjgzZTY5OWViOTUzNTcxMTg2ZWU2MmU3OGVhNTZiZmU1Mjc4OGViMjNiYzgxNzBjYWEyMmQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.2CTbOQxIBVy0dYCBzsuYcls1p9Vca164AqRm_u50W38) ![04g PSP Ident](https://private-user-images.githubusercontent.com/192350063/399485826-af78434a-2990-4e4c-80d7-9abfda4b75bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2NTYzOTcsIm5iZiI6MTczNTY1NjA5NywicGF0aCI6Ii8xOTIzNTAwNjMvMzk5NDg1ODI2LWFmNzg0MzRhLTI5OTAtNGU0Yy04MGQ3LTlhYmZkYTRiNzViZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMVQxNDQxMzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZGEzZTg5YWEwMDQ0OWRjMWU3ZDRkZWYzNzY0ODBiMGVkODA5Y2I2YWZkOGE2YmNmZTI0MTlhYzAyNDJkN2E5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.4GUnswQzrC0htnnE_BL3iNGu9rOPBRA_PWyEjpoyXDY) ![11g PSP Ident](https://private-user-images.githubusercontent.com/192350063/399485836-2af22d70-9f8b-49c0-81a6-b40e8292ebe5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2NTYzOTcsIm5iZiI6MTczNTY1NjA5NywicGF0aCI6Ii8xOTIzNTAwNjMvMzk5NDg1ODM2LTJhZjIyZDcwLTlmOGItNDljMC04MWE2LWI0MGU4MjkyZWJlNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMVQxNDQxMzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zOWNmMWE2MDI3YWY4MzA3OTQxYjE0YTc5ZWYzZTkxMzc1MTk0NGJhNTNhMDRlZWI5YzAwMGU5MDk3N2I4NzlmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.PfzLF5i4LWwVC2QxBtL2ON60cHD8nZSoY7A3SlbNhFg)

I have an 85v1 that works fine. Let me retest with the newest just to verify it though.

MizeryUwU, 2024-12-31:
@krazynez I simply placed 661.pbp on the root of my sd, and copied ARK_DC on PSP/GAME. afterwards, I ran the DC installer and the screen after some time, specifically said that I successfully installed it. I turned off my psp, held left trigger turned it on and poof didn't work.

MizeryUwU, 2024-12-31:
I will try deleting the TM folder and see if it works

krazynez, 2024-12-31:
> @krazynez I simply placed 661.pbp on the root of my sd, and copied ARK_DC on PSP/GAME. afterwards, I ran the DC installer and the screen after some time, specifically said that I successfully installed it. I turned off my psp, held left trigger turned it on and poof didn't work.

Yeah just logic bombs on my part I see I need more coffee lol

krazynez, 2024-12-31:
@MizeryUwU are you on discord? So we don't fill up this GH issue.

MizeryUwU, 2024-12-31:
@krazynez yes I am

MizeryUwU, 2024-12-31:
My name is exactly the same as this handle

krazynez, 2024-12-31:
> My name is exactly the same as this handle

Here just going here and ping my (same name) in #ark-cfw channel https://discord.gg/bePrj9W

lala1818, 2024-12-31:
krazynez 
If I ask you very nicely, can you make a seperate version of new cIPL where the Ltrigger is load MS IPL feature removed please? Then it is will solve my problem. Can you do it for me please?

krazynez, 2024-12-31:
> krazynez If I ask you very nicely, can you make a seperate version of new cIPL where the Ltrigger is load MS IPL feature removed please? Then it is will solve my problem. Can you do it for me please?

I don't know why you would want to do that though that is what we are confused about.

lala1818, 2024-12-31:
krazynez
Just please do it. I ask you very very kindly. Please I'm begging you.

JoseAaronLopezGarcia, 2024-12-31:
L trigger has always been the default option for booting DC. I see no need to change things and confuse users even further.
Unless OP can provide a convincing and well explained reason for this change, I will be closing the thread.

closed 2024-12-31 by JoseAaronLopezGarcia