PSP-Archive/ARK-4 #440: [CL] pressing `square` from `Plugins` and `Settings` resets `CFW Settings`

Aneemasta, 2024-12-22:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

FW 6.60 ARK 4.20.69r151 cIPL

### Homebrew/GAME (if applicable)

Custom Launcher

### Plugins ( if applicable )

game, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, on
game, ms0:/SEPLUGINS/SSSSS/SSSSSBMP.PRX, off
game, ms0:/SEPLUGINS/SSSSS/SSSSSJPG.PRX, off
ULJM05472, ms0:/SEPLUGINS/DIVAPATCH/DIVAPATCH.PRX, on
game, ms0:/SEPLUGINS/CUSHOME/CUSHOME.PRX, on

### Steps to reproduce

1. Launch `CL`
2. Press `Δ` to open menu switcher
3. Select `CFW Settings` and press `X`
4. Set some values to their non-default states
5. Press `Δ` again to open menu switcher
6. Select either `Plugins` or `Settings` menu and press `X`
7. Press `[]`
8. Press `Δ` again to open menu switcher
9. Select `CFW Settings` and press `X`
#### What was expected:
- `CFW Settings` remained unchanged
#### What was happened:
- `CFW Settings` reset to their defaults

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

On

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-12-22:
Your pressing square which resets CFW settings that is expected. I do need to make sure via Plugins menu it doesn't though which thought it didn't but I will double check.

Aneemasta, 2024-12-22:
> Your pressing square which resets CFW settings that is expected
It is from the `CFW Settings`, there is a prompt describing it. What I`m reporting is that pressing `square` from the other 2 menus - `Plugins` and `Settings` (non-CFW) - preforms the same action, related to the `CFW Settings` and unrelated to the other 2.

krazynez, 2024-12-22:
I see thanks,

Fixed in latest pre-release

Aneemasta, 2024-12-23:
> Fixed in latest pre-release

Confirmed, many thanks!

closed 2024-12-23 by Aneemasta