PSP-Archive/ARK-4 #441: [CL/Files] Navigation with `left`/`right` in the context menu doesn't work as expected

Aneemasta, 2024-12-22:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

FW 6.60 ARK 4.20.69r151 cIPL

### Homebrew/GAME (if applicable)

Custom Launcher

### Plugins ( if applicable )

game, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, on
game, ms0:/SEPLUGINS/SSSSS/SSSSSBMP.PRX, off
game, ms0:/SEPLUGINS/SSSSS/SSSSSJPG.PRX, off
ULJM05472, ms0:/SEPLUGINS/DIVAPATCH/DIVAPATCH.PRX, on
game, ms0:/SEPLUGINS/CUSHOME/CUSHOME.PRX, on

### Steps to reproduce

1. Launch `CL`
2. Press `Δ` to open menu switcher
3. Select `Files` and press `X`
4. Press `LB` to open context menu
5. Press `D-Pad Right` repeatedly until reaching the last menu item
6. Press `D-Pad Left` repeatedly until reaching the first menu item
#### What was expected:
- Navigation skipped the same number of items back and forth
#### What was happened:
- First `D-Pad Right` press (from `Cancel`) navigated to non-existing item between `Cut` and `Delete`, thus skipping 2 items
- Next `D-Pad Right` presses navigated to `Create new` and `Go to disc0:/` respectively, thus skipping 2 and 3 items respectively
- First `D-Pad Left` press (`from `Go to disc0:/`) navigated to `Go to `ms0:/`, thus skipping 1 item
- Next `D-Pad Left` press navigated to `Rename`, thus skipping 2 items
- Next `D-Pad Left` presses navigated to `Cut` and `Cancel` respectively, thus skipping 1 and 1 item respectively

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

On

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-12-22:
I can work on this, its because there are hidden options depending on model and if things are in the clipboard that it jumps around like that.

krazynez, 2024-12-22:
There latest release just jumps 0 -> 1/2 -> full and vise versa.

Aneemasta, 2024-12-23:
> There latest release just jumps 0 -> 1/2 -> full and vise versa.

Confirmed, many thanks!

closed 2024-12-23 by Aneemasta