PSP-Archive/ARK-4 #443: [XMB] Unable to launch `CL` and make any changes to `Custom Firmware Settings`

Aneemasta, 2024-12-23:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

FW 6.61 ARK 4.20.69 r151 (v1734905714)

### Steps to reproduce

#### Prerequisites:
- OFW 6.61 installed using https://github.com/krazynez/Chronoswitch and https://darthsternie.net/psp-firmwares/
- `ARK_01234` savedata completely removed from MS (backed up to PC if needed)
#### STR:
- ON PC:
1. Download [ARK-4-v1734905714.zip](https://github.com/PSP-Archive/ARK-4/releases/download/v1734905714/ARK-4-v1734905714.zip)
2. From `ARK-4-v1734905714.zip` unzip `ARK_01234` and `ARK_Loader` to the PSP following [the installation instruction](https://github.com/PSP-Archive/ARK-4/wiki/Installation-on-PSP)
- ON PSP
1. Run `ARK Loader` following the same installation instruction above
2. Wait loading for finish
3. From `XMB` open `Custom Firmware Settings` and alter some of them from their default states
4. From `XMB` run `Custom Launcher`
#### What was expected:
- `CL` run
- `CFW Settings` menu displayed changes from `ON PSP/Step 3`
#### What was happened:
- `XMB` re-run instead
- `Custom Firmware Settings` menu kept defaults
---
P.S.: I've noticed the described behaviour even before full re-installation. Applying `cIPL`, `Full Flash Installation` and `fatms371_mod` changes nothing, we have there CFW settings synced between `CL` and `Recovery` but not `XMB/Custom Firmware Settings`. Most likely the issue is within `XMB Control`.

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-12-25:
Your saying CL icon did not launch CL? I fixed that in a branch but was not aware of it being an issue in main?

Aneemasta, 2024-12-25:
> Your saying CL icon did not launch CL? I fixed that in a branch but was not aware of it being an issue in main?

Very true. To sum up, we currently have:
- inability to launch CL from XMB
- inability to customize CFW settings from XMB
- broken localization of additional XMB items (English only regardless of selected system lang)

krazynez, 2024-12-25:
> Very true. To sum up, we currently have:
>
>     * inability to launch CL from XMB
>
>     * inability to customize CFW settings from XMB
>
>     * broken localization of additional XMB items (English only [...]

Oh okay yeah give me a few days to hopefully merge a branch and that should be all fixed. Was odd I only noticed it when I started the branch not prior, but have not pushed any of that downstream.

Aneemasta, 2024-12-25:
> give me a few days to hopefully merge a branch and that should be all fixed

Looking forward to it. Thanks in advance and merry Xmas!

Aneemasta, 2024-12-29:
@krazynez, check please my latest comment in #444. There seems to be some positive changes related to this issue applied to release mentioned there.

krazynez, 2024-12-29:
> @krazynez, check please my latest comment in #444. There seems to be some positive changes related to this issue applied to release mentioned there.

Correct this is halfway fixed. The second half is we have to check for SETTING.TXT in flash1 like the UPDATE.TXT has.

Aneemasta, 2025-01-31:
Guys, I believe this issue is already solved, isn't it?

closed 2025-01-31 by krazynez