PSP-Archive/ARK-4 #444: [CL/CFW Settings] Resetting via `square` removes `Inferno Cache` exception from `SETTINGS.TXT`

Aneemasta, 2024-12-23:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

FW 6.61 ARK 4.20.69 r151 cIPL v1734905714

### Homebrew/GAME (if applicable)

CL

### Steps to reproduce

#### Prerequisites:
- `Autoboot Launcher` enabled via `Recovery` (due to #443)
- `ms0:/PSP/SAVEDATA/ARK_01234/SETTINGS.TXT` has following strings within:
`# Luxor doesn't like Inferno Cache`
`ULUS10201, infernocache, off`
#### STR:
1. Run `CL`
2. Open `CFW Settings` from menu selector
3. Press `[]` to reset settings
4. Open `Files` from menu selector
5. Navigate to `ms0:/PSP/SAVEDATA/ARK_01234` and open `SETTINGS.TXT`
6. Press `D-Pad Up` to see the end of file
#### What was expected:
- Inferno Cache exception remained untouched
#### What was happened:
- Nothing left after QA flags setting

### USB Charge

Always

### Overclock

Disabled

### Power Save

Launcher

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

On

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-12-28:
Added in latest pre-release.

Aneemasta, 2024-12-29:
> Added in latest pre-release.

[v1735398053](https://github.com/PSP-Archive/ARK-4/releases/tag/v1735398053): Confirmed. Some weird thing occurred though:
1. Attempt to update via `UPDATE` resulted in PSP crashing.
2. Attempt to update via `ARK_Launcher` was a success.
3. When rebooting after re-`Full Flash Installation` all of `CFW Settings` was reset to none/off/disable. This hadn't been happening after.
4. The issue #443 seems to be gone too, but I can't be entirely sure.

krazynez, 2024-12-29:
> > Added in latest pre-release.
> 
> [v1735398053](https://github.com/PSP-Archive/ARK-4/releases/tag/v1735398053): Confirmed. Some weird thing occurred though:
> 
>     1. Attempt to update via `UPDATE` resulted in PSP crashing.
> 
>     2. Attempt to update via `ARK_Launcher` was a success.
> 
>     3. When rebooting after re-`Full Flash Installation` all of `CFW Settings` was reset to none/off/disable. This hadn't been happening after.
> 
>     4. The issue [[XMB] Unable to launch `CL` and make any changes to `Custom Firmware Settings` #443](https://github.com/PSP-Archive/ARK-4/issues/443) seems to be gone too, but I can't be entirely sure.

So FFI (Full Flash Install) generates new files so it would ovewrite previous files. FFI should be done after Live its how it is in the wiki but not clearly stated enough. I will checkout the ARK-4 Update on xmb.

Aneemasta, 2024-12-29:
>I will checkout the ARK-4 Update on xmb.

Very well. Currently the issue reported here is solved. Shall we close it or shall we keep it opened a bit more as a reminder of `UPDATE` malfunction?

krazynez, 2024-12-29:
> > I will checkout the ARK-4 Update on xmb.
> 
> Very well. Currently the issue reported here is solved. Shall we close it or shall we keep it opened a bit more as a reminder of `UPDATE` malfunction?

we can close it once I get out of bed I will work on that. I have other ARK stuff I need to work onnas well.

Aneemasta, 2024-12-29:
Fine by me.
I'll keep conducting exploratory testing then. Stay tuned for further reports.

closed 2024-12-31 by Aneemasta