PSP-Archive/ARK-4 #448: Custom Launcher Crashes

NikeDudeeGAMER, 2024-12-28:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

v1735398053

### Homebrew/GAME (if applicable)

Custom Launcher

### Plugins ( if applicable )

always, ms0:/SEPLUGINS/npdrm_free.prx, on

xmb, ms0:/SEPLUGINS/dayviewer.prx, on

xmb, ms0:/SEPLUGINS/category_lite.prx, on

### Steps to reproduce

when i turn on the custom launcher the psp hd light blinks a few times then it turns off the psp

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-12-28:
Gonna need a bit more here. Can't reproduce this.

NikeDudeeGAMER, 2024-12-28:
> Gonna need a bit more here. Can't reproduce this.

what more do i give you

krazynez, 2024-12-28:
> > Gonna need a bit more here. Can't reproduce this.
> 
> what more do i give you

Do you have any games with corruoted icons? That has been an issue in the past.

krazynez, 2024-12-28:
Recopying ARK_01234 fixed ( solved in discord )

closed 2024-12-28 by krazynez