PSP-Archive/ARK-4 #453: Advanced Recovery Bug (POWER+L)

WeiKai18, 2025-01-01:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

ARK 4.20.69 r151

### Plugins ( if applicable )

VSH, ms0:/seplugins/cxmb/cxmb.prx

### Steps to reproduce

1.) Enable CXMB plugin 
2.) Power off device
3.) Hold L+Power to get into advanced recovery
4.) Advanced recovery will pop up for about 3 seconds before PSP switches off and then switches on and starts the PSP intro before turning off again and constantly doing this in a loop

NOTE - I have tried other plugins and they work fine, i can get into advanced recovery and i can stay there without bootlooping. (WITH CXMB PLUGIN DISABLED)

I have also tried to isolate CXMB on its own and it still produces the same issue when i enable the plugin. I have also tried to enable the plugin and stay on the default PSP Theme but the issue persists.

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-02-04:
Any idea if this has been fixed?

WeiKai18, 2025-02-08:
Unfortunately not

JoseAaronLopezGarcia, 2025-03-16:
Fixed in latest (pre)release.

closed 2025-03-16 by JoseAaronLopezGarcia