PSP-Archive/ARK-4 #454: ARK-4 cannot start games in Adrenaline 7.1.3

0lorak, 2025-01-01:
### PSP/Vita Model

VTE-1000 (Vita TV)

### ARK Version

4.20.69 r151

### Homebrew/GAME (if applicable)

Any iso, it seems to work ok for homebrew though.

### Plugins ( if applicable )

No plugins.

### Steps to reproduce

1. Launch [Adrenaline 7.1.3](https://github.com/isage/Adrenaline/releases)
2. Launch ARK Loader. ARK Loader is in /PSP/GAME and ARK_01234 in /PSP/SAVEDATA/. I always launching ARK-4 like that on my PS Vita. 
3. Start any iso. For homebrews it seems to work with no issues.
4. The system frezees and then crash at the PSP boot image showing this error code C1-2858-3.

### USB Charge

Disabled

### Overclock

Always

### Power Save

XMB

### Balanced Energy Mode

Pops

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

isage, 2025-01-02:
If i might add, @0lorak said it started on 7.1.2 (maybe earlier)
And the only huge diff i can think of is using up-to-date pspsdk to build adrenaline.

0lorak, 2025-01-03:
Yes, ARK-4 works with no issues in Adrenaline 7 but cannot start PSP isos or PS1 games from Adrenaline 7.1.2 until the current one Adrenaline 7.1.4. I have no tests yet on Adrenaline 7.1.0 and 7.1.1 though.

isage, 2025-01-11:
Tracked down to https://github.com/isage/Adrenaline/compare/v7.1.1...v7.1.2#diff-1ba641a94c1d3d3f3a0bc91300362f3f11c3e68f94747cde226adcf00d2deda1R519

Issue is that short names are required for xmb photo/video browser to work properly with camera images/videos, so i can't really revert it

JoseAaronLopezGarcia, 2025-01-11:
It's probably conflicting with something here:
https://github.com/PSP-Archive/ARK-4/blob/main/core/stargate/chn_iso.c

Or here:
https://github.com/PSP-Archive/ARK-4/blob/890c4bbe42fc9f52e4e32b1bb08b350f1cb850e8/core/vshctrl/virtual_pbp.c#L1014

0lorak, 2025-01-15:
As an extra comment, ARK-4 cannot be started on the newest Adrenaline 7.1.6. This happens since Adrenaline 7.1.5 by the way.

isage, 2025-01-16:
Works fine for me through ARK Loader, (although older ARK version, 4.20.34)

JoseAaronLopezGarcia, 2025-01-17:
This should work: https://github.com/PSP-Archive/ARK-4/releases/tag/v1737136031

0lorak, 2025-01-18:
> This should work: https://github.com/PSP-Archive/ARK-4/releases/tag/v1737136031

@JoseAaronLopezGarcia The ARK-4 version (4.20.69 r151) you mention above starts on Adrenaline 7.1.6. Although, I am getting error 80010002 when trying to launch a PS1 game and a freeze (error code C1-2858-3) on PSP game boot animation, similar to "Tony Hawk's Underground 2" referred above, when trying to start a PSP iso on that version of ARK-4. Same situation on ARK-4 (4.20.69 r153).

Both, PSP isos and PS1 games start with no issues on Adrenaline 7.1.6 though.

pmsobrado, 2025-01-19:
> As an extra comment, ARK-4 cannot be started on the newest Adrenaline 7.1.6. This happens since Adrenaline 7.1.5 by the way.

You mean it gets stuck in "loading stage 2" message?

0lorak, 2025-01-20:
> > As an extra comment, ARK-4 cannot be started on the newest Adrenaline 7.1.6. This happens since Adrenaline 7.1.5 by the way.
> 
> You mean it gets stuck in "loading stage 2" message?

Yes. This problem has been fixed on the latest ARK-4 version.

pmsobrado, 2025-01-20:
> > > As an extra comment, ARK-4 cannot be started on the newest Adrenaline 7.1.6. This happens since Adrenaline 7.1.5 by the way.
> > 
> > 
> > You mean it gets stuck in "loading stage 2" message?
> 
> Yes. This problem has been fixed on the latest ARK-4 version.

You mean latest builds but not latest official release right?

0lorak, 2025-01-20:
> You mean latest builds but not latest official release right?
Yes, [this one.
](https://github.com/PSP-Archive/ARK-4/releases/tag/v1737151905)

pmsobrado, 2025-01-20:
> > You mean latest builds but not latest official release right?
> > Yes, [this one.
> > ](https://github.com/PSP-Archive/ARK-4/releases/tag/v1737151905)

Thanks!

krazynez, 2025-02-04:
Good to close?

JoseAaronLopezGarcia, 2025-02-04:
Fixed.

closed 2025-02-04 by JoseAaronLopezGarcia