PSP-Archive/ARK-4 #455: Hide/move CFW items in VSH

ThatOneBonk, 2025-01-03:
### New Feature

Upon installing ARK-4, new XMB elements are added to the Extras/Game sections of the menu. Personally, I don't want the custom launcher, neither do I want the plugins manager on my XMB. There should be a way to either: 
- hide these elements, perhaps in CFW Settings -> Hide Menu Items [Custom Launcher/Plugins/Both], or
- move these elements somewhere else, perhaps in CFW Settings -> ARK-4 Options Location [Settings Tab/Extras Tab/Game Tab].

JoseAaronLopezGarcia, 2025-01-03:
Delete XMBCTRL.PRX in ARK_1234.

ThatOneBonk, 2025-01-03:
This does absolutely nothing on a permanent install of ARK-4.

ThatOneBonk, 2025-01-04:
I'm sure this will work, and thank you for the advice, but I believe that I shouldn't have to do that. 
I believe that there should be a user-friendly way to disable or change XMB features that they don't want *after* installing the CFW, so this feature request is still valid, especially the "move the elements" part. I wouldn't mind if the features were in the Settings tab, for example. Not sure if this is possible to implement or not, though.

ThatOneBonk, 2025-01-04:
The Category Lite plugin adds its options into the Systems category in the Settings tab. Surely, ARK-4 can do this as well, if a dedicated category isn't an option?

wad11656, 2025-02-04:
Hear hear! All the wonky unrelated icons added are a bit gross 🫣

JoseAaronLopezGarcia, 2025-02-04:
Just delete XMBCTRL.PRX from ARK_01234, there's really not more to it.

JoseAaronLopezGarcia, 2025-02-04:
If you get rid of the extras categories then CFW stuff will be in the Game section.

closed 2025-03-16 by JoseAaronLopezGarcia