PSP-Archive/ARK-4 #457: [UPDATE] Crash while updating via `ms0:/PSP/GAME/UPDATE` with `fatms371_mod` installed

Aneemasta, 2025-01-04:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

6.61 ARK 4.20.69 r151 cIPL

### Homebrew/GAME (if applicable)

UPDATE

### Plugins ( if applicable )

always, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, on
always, ms0:/SEPLUGINS/SSSSS/SSSSSJPG.PRX, off
always, ms0:/SEPLUGINS/SSSSS/SSSSSBMP.PRX, off
ULJM05472, ms0:/SEPLUGINS/DIVAPATCH/DIVAPATCH.PRX, on
game, ms0:/SEPLUGINS/CUSHOME/CUSHOME.PRX, on

### Steps to reproduce

#### Prerequisites:
- `fatms371_mod` installed from `ARK Full Installer`
#### STR:
1. Launch `UPDATE` any preferable way:
- From `XMB\Settings\ARK-4 Updater`
- From `XMB\Game\Memory Stick\ARK UPDATE`
- From `CL\Games\UPDATE`
#### What was expected:
- Update was applied with no problem
#### What was happened:
- PSP crashes

P.S.: @krazynez, this is the same problem mentioned earlier in #444. Not sure if my PSP gains anything from using `fatms371_mod` and whether I should use it at all, but a bug is a bug.

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

On

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2025-01-04:
I just did this and it worked fine. Perhaps its your game plugins?

Aneemasta, 2025-01-05:
Now that's strange.
OK, I've tried a complete reinstalling with Chronoswitch and OEM FW 6.61. Here is what was done:
1. Downgraded to OEM 6.61 using Chronoswitch
2. Removed all of ARK files from MS: savedata and `PLUGINS.TXT` from `SEPLUGINS`
(at this point my PSP should have not a single thing left deployed from ARK)
3. Unzipped brand new savedata, loader, cIPL flasher and full installer from [latest "stable" release](https://github.com/PSP-Archive/ARK-4/releases/tag/r42069-151)
4. Loaded ARK using loader
5. Flashed cIPL using flasher
6. Performed FFI using full installer
7. Installed `fatms371_mod` using full installer
8. Unzipped `UPDATE` from [latest rolling release](https://github.com/PSP-Archive/ARK-4/releases/tag/v1736023245)
9. Started update process from `XMB/Settings/ARK-4 Updater`
These steps resulted in PSP went crash, so I doubt the problem is with plugins (unless their presence in `SEPLUGINS` subfolders without adding to `PLUGINS.TXT` affects as well?).
And, when I uninstalled `fatms371_mod` using the same full installer (from stable release) and tried to update the same way, it went perfectly fine.
Worth mention that MS my PSP use isn't original, it's just random Kingston 128Gb microSD connected via random mSD-to-MS adapter. I do have the original Sony MS and gonna try to reproduce this big with it as soon as I'll come back from trip.
---
Summary, I have several possible things conflicting when combined:
- `fatms371_mod`
- Unoriginal MS
- `Full Flash Installing`
- `Custom IPL`

krazynez, 2025-01-10:
Yeah I see it now.

JoseAaronLopezGarcia, 2025-02-04:
I'm gonna close this for a very simple reason: we did not make fatmsmod and we won't be maintaining it.
Sadly it's known to cause a lot of issues. You could try with things like disabling "Memory Stick Speedup", perhaps it'll help.

closed 2025-02-04 by JoseAaronLopezGarcia

Aneemasta, 2025-02-09:
> I'm gonna close this for a very simple reason: we did not make fatmsmod and we won't be maintaining it. Sadly it's known to cause a lot of issues. You could try with things like disabling "Memory Stick Speedup", perhaps it'll help.

Sad but true. Well, I still has the option to remove fatms mod before update and restore it after.
Also, this bug probably worth mentioning as a known issue on the ["Update" wiki page](https://github.com/PSP-Archive/ARK-4/wiki/Update).