PSP-Archive/ARK-4 #459: Tony Hawk's Underground 2 freezing on PSP Game boot animation on PS Vita with ARK-4

Haasman0, 2025-01-05:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

 r42069-151

### Homebrew/GAME (if applicable)

ULUS10014

### Plugins ( if applicable )

None

### Steps to reproduce

When I launch ARK-4 then it boots me back to the XMB and when I choose THU2 then it shows the PSP boot animation then just freezes at the end of the animation and I have to force shutdown my PS Vita.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-01-17:
Could this be related to this other report? https://github.com/PSP-Archive/ARK-4/issues/454

0lorak, 2025-01-18:
> Could this be related to this other report? [#454](https://github.com/PSP-Archive/ARK-4/issues/454)

I think so, as this report shows that it happens on a PS Vita too.

JoseAaronLopezGarcia, 2025-01-24:
Was this fixed with latest release? I can't seem to reproduce it myself, works fine here.

closed 2025-02-04 by JoseAaronLopezGarcia