PSP-Archive/ARK-4 #461: PSP ISOs crash on boot after updating to r152

MizeryUwU, 2025-01-07:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

Version 4.20.69 r152

### Plugins ( if applicable )

game, ms0:/seplugins/cwcheat/cwcheat.prx, on
game, ms0:/seplugins/battery_icon.prx, on
game, ms0:/seplugins/bbsfmUndub.prx, on
game, ms0:/seplugins/macrofire.prx, on
vsh, ms0:/seplugins/categorieslite/category_lite.prx, on
vsh, ms0:/seplugins/battery_icon.prx, on
vsh, ms0:/seplugins/hold.prx, on
vsh, ms0:/seplugins/cxmb/cxmb.prx, on
vsh, ms0:/seplugins/dayviewer.prx, on
pops, ms0:/seplugins/cwcheat/cwcheatpops.prx, on
pops, ms0:/seplugins/battery_icon.prx, on

### Steps to reproduce

1. Updated ARK 4 through settings>ARK 4 Updater
2. CFW successfully updated

Consequence:
1. PSP ISOs (games) crash on PSP logo boot screen.
2. Homebrews boot just fine
3. PS1 Games boot just fine

Actions taken:
1. Reverted to r151

Result:
1. Everything is back to Normal 

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Homebrew

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-01-07:
I can confirm it is indeed a bug.

krazynez, 2025-01-08:
> I can confirm it is indeed a bug.
> 
> Edit: nevermind, I couldn't reproduce it, my crash was unrelated (bad plugin). I can boot ISO just fine in latest build.

Same, I boot ISO's with no issues.

MizeryUwU, 2025-01-08:
Now it says the game could.not be started 80020321

MizeryUwU, 2025-01-08:
So I think I will be staying in r151. I don't see any reason why it would work on my psp except r152 where nothing has really changed. Maybe it has something to do with plugins again having some sort of compatibility issues with ark 4 just like dcark which is a shame really.

JoseAaronLopezGarcia, 2025-01-08:
> So I think I will be staying in r151. I don't see any reason why it would work on my psp except r152 where nothing has really changed. Maybe it has something to do with plugins again having some sort of compatibility issues with ark 4 just like dcark which is a shame really.

which plugins do you have installed?

MizeryUwU, 2025-01-08:
> which plugins do you have installed?

I stated it above under plugins

MizeryUwU, 2025-01-08:
Last issue I encountered on r151 is that I can't boot DCARK recovery menu with L+power while categories lite and hold plugins are enabled. You have to manually disable them on. TXT to boot that menu.

JoseAaronLopezGarcia, 2025-01-08:
> Last issue I encountered on r151 is that I can't boot DCARK recovery menu with L+power while categories lite and hold plugins are enabled. You have to manually disable them on. TXT to boot that menu.

That bug should have been fixed in r152

MizeryUwU, 2025-01-08:
Hold on ill check

MizeryUwU, 2025-01-08:
> > So I think I will be staying in r151. I don't see any reason why it would work on my psp except r152 where nothing has really changed. Maybe it has something to do with plugins again having some sort of compatibility issues with ark 4 just like dcark which is a shame really.
> 
> which plugins do you have installed?
> 
> edit: nevermind I see it's in the report
> 
> Can you try running it without any plugins on the xmb?

Yeah. It's confirmed. Another plugin compatibility issue. I disabled all plugins and now I have been able to boot the ISOs just fine.

MizeryUwU, 2025-01-08:
IL have to check which plugin causes this issue though

JoseAaronLopezGarcia, 2025-01-08:
I did change the way plugins loaded thinking it would fix an issue with cheatdevice_remastered, but it turns out the issue was somewhere else and fixed by the plugin itself, but I totally forgot to revert back to how it was before.

I will make a build and post it here for you to test it.

MizeryUwU, 2025-01-08:
> I did change the way plugins loaded thinking it would fix an issue with cheatdevice_remastered, but it turns out the issue was somewhere else and fixed by the plugin itself, but I totally forgot to revert back to how it was before.
> 
> I will make a build and post it here for you to test it.

Okay thank you

MizeryUwU, 2025-01-08:
> IL have to check which plugin causes this issue though

So apparently vsh and pops related plugins are good except game-related plugins.

JoseAaronLopezGarcia, 2025-01-08:
[FLASH0.zip](https://github.com/user-attachments/files/18348517/FLASH0.zip)

Extract FLASH0.ARK and copy it to /PSP/SAVEDATA/ARK_01234/ (overwrite), fully restart the console while holding Home so you are back to stock firmware and run ARK Loader again (this will re-install flash0 archive).

Let me know if that fixes it.

MizeryUwU, 2025-01-08:
> [FLASH0.zip](https://github.com/user-attachments/files/18348517/FLASH0.zip)
>
> Extract FLASH0.ARK and copy it to /PSP/SAVEDATA/ARK_01234/ (overwrite), fully restart the console while holding Home so [...]
Nice! It actually fixed the issue man.

JoseAaronLopezGarcia, 2025-01-08:
Awesome, pushed to repo.

MizeryUwU, 2025-01-08:
Thank you kind sir. L+power is also now booting Dc ark menu without any hiccups. Now it seems all plugin-related compatibility issues are fixed. Cheers to the ark 4 dev team.