PSP-Archive/ARK-4 #463: Multiple psx games stopped working

rbzikadze, 2025-01-11:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r153

### Steps to reproduce

Some psx2psp games that worked before recent updates fail to launch, not sure which update broke them
1. Try to launch CTR, Hercules, Pandemonium (games that worked before)
2. Either black screen and reboot or error 80010002 depending on extra memory enabled or not
3. Issue persists on ps vita as well as psp2000

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Homebrew

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-01-11:
Doesn't really make any sense, ARK doesn't determine what PS1 games are compatible. That's an official Sony emulator.
Perhaps it's an issue with plugins, are you sure you're running the latest version of ARK? We had to fix a regression with plugins.

rbzikadze, 2025-01-11:
I've tried multiple times with plugins enabled and disabled, games worked before with those enabled, they don't launch from xmb as well as ark menu. I can provide more details if necessary, f.e CTR game id is NPUI94426, Hercules is NPUJ01029. Both games have been working before

JoseAaronLopezGarcia, 2025-01-11:
We haven't touched anything related to PS1 games in a long time, but it's probably some other unrelated code causing issues.
If you could try pre-releases until you figure out which one broke those games then we can pinpoint the faulty code.

nepulaw, 2025-01-15:
i had a same issue on my psp 1000 model and i think it has to do with the icon for the game in psp stxe and not psn ps1 size as the ones with the psn ps1 size work but not the other ones which is quite odd

JoseAaronLopezGarcia, 2025-01-17:
Any news on this? It would be idea if we could have an report on the specific pre-release that started causing issues.

JoseAaronLopezGarcia, 2025-01-23:
I will be closing this report as there hasn't been an update regarding the situation of this issue with the latest pre-release and I cannot confirm the issue myself since all my PS1 games are still working fine.

Feel free to re-open it if you consider it necessary.

closed 2025-01-23 by JoseAaronLopezGarcia