PSP-Archive/ARK-4 #464: Update from 6.60 PRO B-10 to ARK-4 results in a softbrick

TAbdiukov, 2025-01-19:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

r153

### Plugins ( if applicable )

As per 6.60 PRO B10,

VSH.TXT,

```
ms0:/cxmb/cxmb.prx 1
ms0:/SEplugins/hold.prx 0
ms0:/SEplugins/hud.prx 0
ms0:/SEplugins/joysens.prx 0
ms0:/SEplugins/joysens_lite.prx 0
ms0:/SEplugins/mp3play.prx 0
ms0:/SEplugins/mp3play_lite.prx 0
ms0:/SEplugins/dayviewer.prx 1
ms0:/SEplugins/category_lite.prx 0
```

GAME.TXT

```
ms0:/SEplugins/vcscheatdevice.prx 0
ms0:/SEplugins/vcscheatdevice371.prx 0
ms0:/SEplugins/hold.prx 0
ms0:/SEplugins/hud.prx 0
ms0:/SEplugins/joysens.prx 0
ms0:/SEplugins/joysens_lite.prx 0
ms0:/SEplugins/lcscheatdevice.prx 0
ms0:/SEplugins/mp3play.prx 0
ms0:/SEplugins/mp3play_lite.prx 0
ms0:/SEplugins/psnlover.prx 0
ms0:/SEplugins/psp3d.prx 0
ms0:/seplugins/cwcheat/cwcheat.prx 0
ms0:/seplugins/AdhocToUSB.prx 0
```

POPS.TXT
```
ms0:/SEplugins/hold.prx 0
ms0:/SEplugins/hud.prx 0
ms0:/SEplugins/joysens.prx 0
ms0:/SEplugins/joysens_lite.prx 0
ms0:/SEplugins/mp3play.prx 0
ms0:/SEplugins/mp3play_lite.prx 0
ms0:/SEplugins/popsloader.prx 0
ms0:/SEplugins/prxshot.prx 0
ms0:/seplugins/cwcheat/cwcheatpops.prx 1
```

### Steps to reproduce

1. Have 6.60 PRO B10, and push the [https://github.com/PSP-Archive/ARK-4/releases/tag/r152](r152) files to their corresponding folders (ms0:/PSP/GAME and ms0:/PSP/SAVEDATA)
2. Per first part of [this guide](https://www.pspunk.com/psp-cfw/), run ARK_Loader , let it do its thing
3. Then I opted for [full flash installation](https://github.com/PSP-Archive/ARK-4/wiki/Full-Flash-Installation), I ran full installer. It detected "pro". I had it perform installation
4. Reboot

Afterwards, I got a softbrick. PSP can access old recovery menu and VSH, but 

* Old VSH select menu
* System Settings - still shows 6.60 PRO B10 software and Mac address hidden with B10 message (not ARK-4 message)
* PSP cannot run any games (either from `ms0://PSP/GAME` or `ms0://ISO`) - everything results in crash without an error message
* PSP cannot start 6.60 Digital Comics
* PSP can try to start `/PSP/GAME/RECOVERY/EBOOT.PBP` from recovery, but it also results in a crash
* Removing/renaming `seplugins` directory (so that plugins have no effect) does not change anything above

Luckily, I worked around the problem by doing the following
1. Starting VSH with SELECT being Held - which successfully resulted in booting into official-like firmware
2. Ran official 661.PBP update - success
3. Updated successfully
4. Then, I installed ARK-4

However, I wanted to notify about issues when working with 6.60 PRO B-10

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-01-19:
You need to [Chronoswitch](https://github.com/PSP-Archive/Chronoswitch) to revert back to OFW first.

TAbdiukov, 2025-01-19:
@krazynez I'm not so sure, because ARK-4 successfully detects "pro" environment, so it is probably designed to be able to overwrite other CFW?

krazynez, 2025-01-19:
No, we only detect PRO/ME if using cIPL.

TAbdiukov, 2025-01-20:
I see, in this case, if this is true, then this is a non-issue. Closing...

closed 2025-01-20 by TAbdiukov

chenweiyj, 2025-02-11:
I have met the same problem. It cannot open any application in /PSP/GAME/ folder, even the OFW update file and Chronoswitch. It is bricked.

chenweiyj, 2025-02-11:
@TAbdiukov  Trying starting your PSP with SELECT button held has no effect for me. It shows the same effect with only push up the power button.

chenweiyj, 2025-02-11:
@TAbdiukov My psp can show pro-b10 vsh menu by press SELECT button, but recovery is not work.

TAbdiukov, 2025-02-11:
@chenweiyj Nonono, I think you misunderstood

1. Completely turn off PSP - remove battery
2. Put battery back
3. Turn PSP on using a power slider
4. Hold SELECT, while it is starting

It should start in official firmware-like mode. Then run the official 6.21 update from the XMB menu

chenweiyj, 2025-02-11:
My PSP is PSP3000, is it the same?

chenweiyj, 2025-02-11:
@TAbdiukov Fixed to recovery to OFW by combination of HOME + O + X + START + SELECT button while booting the system. Thank you.

JoseAaronLopezGarcia, 2025-02-11:
> [@TAbdiukov](https://github.com/TAbdiukov) Fixed to recovery to OFW by combination of HOME + O + X + START + SELECT button while booting the system. Thank you.

HOME alone is enough, I don't know where you got the other buttons from. Next time take the extra effort of reading the Wiki, button combos are explained in the main page.