PSP-Archive/ARK-4 #465: Crash when attempting to access Recovery Menu from ARK when launched from Adrenaline

pmsobrado, 2025-01-19:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.69 r153

### Plugins ( if applicable )

None.

### Steps to reproduce

For some reason recovery menu is not working (ARK crashes before showing it) when ARK is inside Adrenaline. It works fine in ARK standalone. I've tried official v7 Adrenaline and also latest builds by isage. Maybe it's normal but I thought I could ask here. Thanks in advance.

1. Launch Adrenaline
2. Launch ARK Loader
3. Press "Select"
4. Select "Recovery menu"

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-01-22:
Update on this: I could confirm that it does indeed happen, I just couldn't figure out what's causing it.

Recovery Menu is pretty redundant in ARK anyways, you can achieve the same result by holding Start+Select when loading the XMB or the Custom Launcher, and just using that to change plugins and settings, which is basically all Recovery Menu ever does anyways.

JoseAaronLopezGarcia, 2025-01-22:
Ok that I didn't know. I will do some more tests.

MizeryUwU, 2025-01-22:
> Update on this: I could confirm that it does indeed happen, I just couldn't figure out what's causing it.
>
> Recovery Menu is pretty redundant in ARK anyways, you can achieve the same result by [...]

The recovery menu is sooooo helpful specially if your psp won't boot let's say, due to a bad plugin without having the need to manually disable it in the plugin.txt. You just use the combination keys on boot to open the recovery menu, turn off bad plugin and boom done, we can your psp boots again.

MizeryUwU, 2025-01-22:
> > > Update on this: I could confirm that it does indeed happen, I just couldn't figure out what's causing it.
> > > Recovery Menu is pretty redundant in ARK anyways, you can achieve the same result by holding Start+Select when loading the XMB or the Custom Launcher, and just using that to change plugins and settings, which is basically all Recovery Menu ever does anyways.
> > 
> > 
> > The recovery menu is sooooo helpful specially if your psp won't boot let's say, due to a bad plugin without having the need to manually disable it in the plugin.txt. You just use the combination keys on boot to open the recovery menu, turn off bad plugin and boom done, your psp boots again. Really really useful specially if your experimenting on plugins.
> 
> If your PSP doesn't boot because of a bad plugin, just hold Start button while powering it on, or while booting a game, and all plugins will be temporarily disabled. This feature is ARK-only and pretty much makes recovery menu redundant. The only time Recovery Menu would be helpful is if you actually cause damages to the XMB (vsh_main.prx) on flash0.

Well yeah, but sometimes it is a hassle to turn on your plugins one by one again especially if you know which plugin is actually disfunctional.

JoseAaronLopezGarcia, 2025-01-22:
Don't worry, we'll fix it, we just don't consider it "high priority" right now.

JoseAaronLopezGarcia, 2025-01-23:
This has been fixed in latest release and pre-release.