PSP-Archive/ARK-4 #469: Psp fat (1000) not showing .ctf themes even with cxmb

Art-doge, 2025-01-27:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK 4.20.69 r152

### Plugins ( if applicable )

VSH, ms0:/SEPLUGINS/cxmb.prx, 1

### Steps to reproduce

1.Download cxmb from pspunk
2.extract and cut cxmb.prx from the cxmb folder
3.connect psp and paste in SEPLUGINS in psp root.
4.Download .ctf theme from pspunk.
5.paste in themes folder in psp/themes.
6.disconnect psp and check in theme settings.
7.does not shoe,but .ptf themes show up just fine.

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-01-27:
cxmb is a bit picky sometimes with folder placement. Remove the plugin and place `cxmb.prx` in `ms0:/SEPLUGINS/cxmb/`

Also after you install a plugin make sure you rrset the PSP for it to take effect.

krazynez, 2025-02-03:
Any update on this?

Art-doge, 2025-02-03:
Sorry for the late reply,I didn't see the email and don't use guthub often. I'll let you know within a day for sure.

Art-doge, 2025-02-03:
> Any update on this?

You mean like this ?

[image]
[image]

Art-doge, 2025-02-03:
> Any update on this?

Aye ! Thanks man,it worked ! Don't know what I would've done without ya.

krazynez, 2025-02-03:
> > Any update on this?
> 
> Aye ! Thanks man,it worked ! Don't know what I would've done without ya.

Good. Closing then.

closed 2025-02-03 by krazynez