PSP-Archive/ARK-4 #472: Psp switches off when enabling ctf instead of reseting vsh

Karim0786, 2025-01-30:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r153

### Homebrew/GAME (if applicable)

Vsh

### Plugins ( if applicable )

vsh,ms0:/seplugins/category_lite.prx On
game.ms0:/SEPLUGINS/TEMPAR.PRX
ps1.ms0:/SEPLUGINS/TEMPAR.PRX 
game.ms0:/SEPLUGINS/npdrm_free_mod.prx
ps1.ms0:/SEPLUGINS/npdrm_free_mod.prx
xmb.ms0:/SEPLUGINS/CXMB/CXMB.PRX

### Steps to reproduce

Idk how to

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-01-30:
Umm what is going on with your plugins listed above? Or is that just type-o's?

Karim0786, 2025-01-30:
Thats the exact spelling on the psp

Karim0786, 2025-01-30:
One i start up the psp again the theme shows

krazynez, 2025-01-30:
> Thats the exact spelling on the psp

that is not correct syntax.

Karim0786, 2025-01-30:
Is that the problem

Karim0786, 2025-01-30:
[image]
[image]
[image]
[image]
[image]
[image]

Karim0786, 2025-01-30:
Trying to upload vid not working

JoseAaronLopezGarcia, 2025-01-31:
Did you make sure that the CTF themes you downloaded are for 6.60/6.61? Themes for older firmwares may not work well if at all.

Karim0786, 2025-02-01:
I made my own theme and it the ctf never reset vsh just switched off

krazynez, 2025-02-03:
> I made my own theme and it the ctf never reset vsh just switched off

it might be your theme then, try another theme from pspunk to verify it does the same thing or not please.

JoseAaronLopezGarcia, 2025-03-16:
Closing as we can't dupplicate.

closed 2025-03-16 by JoseAaronLopezGarcia