PSP-Archive/ARK-4 #473: Polish Translation

Minelaks1, 2025-01-30:
### New Feature

`{
    "Game": "Gra",
    "Games": "Gry",
    "Files": "Pliki",
    "Network": "Sieć",
    "Settings": "Ustawienia",

    "UMD Drive": "Napęd UMD",
    "Recovery Menu": "Menu odzyskiwania",

    "Exit": "Wyjście",
    "Restart": "Restart",
    "Shutdown": "Wyłącz",

    "Disabled": "Wyłączone",
    "Enabled": "Włączone",
    "Confirm": "Potwierdź",
    "Default": "Domyślne",
    "Remove": "Usuń",
    "None": "Brak",

    "Loading games...": "Ładowanie gier...",
    "No games available": "Brak dostępnych gier",
    "Game Options": "Opcje gry",
    "View Info": "Zobacz informacje",
    "New name for Game": "Nowa nazwa gry",
    "Game Info": "Informacje o grze",
    "Name - ": "Nazwa - ",
    "Game ID - ": "ID gry - ",
    "Path - ": "Ścieżka - ",
    "Size - ": "Rozmiar - ",

    "Folder": "Folder",
    "File": "Plik",
    "Cancel": "Anuluj",
    "Copy": "Kopiuj",
    "Cut": "Wytnij",
    "Paste": "Wklej",
    "Delete": "Usuń",
    "Rename": "Zmień nazwę",
    "Create New": "Utwórz nowy",
    "Go to ms0:/": "Przejdź do ms0:/",
    "Go to ef0:/": "Przejdź do ef0:/",
    "Go to ftp:/": "Przejdź do ftp:/",
    "Go to disc0:/": "Przejdź do disc0:/",
    "<refresh>": "<odśwież>",
    "<disconnect>": "<rozłącz>",
    "Install Theme": "Zainstaluj motyw",
    "Preview": "Podgląd",
    "Install": "Instaluj",
    "Accept Install": "Potwierdź instalację",
    "Use theme without installing": "Użyj motywu bez instalacji",
    "Install Plugin": "Zainstaluj wtyczkę",
    "<Game ID>": "<ID gry>",
    "Game ID (i.e. ULUS01234)": "ID gry (np. ULUS01234)",
    "Install path": "Ścieżka instalacji",
    "Installing Plugin": "Instalowanie wtyczki",
    "Extract here": "Wypakuj tutaj",
    "Extract to ": "Wypakuj do ",
    "Extract Archive to": "Wypakuj archiwum do",
    "Extracting archive": "Rozpakowywanie archiwum",
    "into": "do",
    "Deleting folder": "Usuwanie folderu",
    "Deleting file": "Usuwanie pliku",
    "Moving file/folder": "Przenoszenie pliku/folderu",
    "Please Wait": "Proszę czekać",
    "Destination exists, what to do?": "Cel już istnieje, co zrobić?",
    "Rename destination": "Zmień nazwę celu",
    "Overwrite": "Nadpisz",
    "Copying file": "Kopiowanie pliku",
    "New name for file/folder": "Nowa nazwa pliku/folderu",
    "Confirm Deletion?": "Potwierdzić usunięcie?",
    "Create New...": "Utwórz nowy...",
    "Name of new folder": "Nazwa nowego folderu",
    "new folder": "Nowy folder",
    "Name of new file": "Nazwa nowego pliku",
    "new file": "Nowy plik",
    "Free size": "Wolne miejsce",

    "Network Tools": "Narzędzia sieciowe",
    "Press [] to check for Updates": "Naciśnij [], aby sprawdzić aktualizacje",
    "FTP Server is stopped. Press %s to start.": "Serwer FTP zatrzymany. Naciśnij %s, aby uruchomić.",
    "FTP Server is running @ %s. Press %s to stop.": "Serwer FTP działa @ %s. Naciśnij %s, aby zatrzymać.",
    "Could not get IP address": "Nie można uzyskać adresu IP",
    "Could not connect to Access Point": "Nie można połączyć się z punktem dostępu",
    "Could not initialize network": "Nie można zainicjalizować sieci",
    "Disconnecting FTP server": "Rozłączanie serwera FTP",
    "FTP Disconnected": "FTP rozłączony",
    "Checking for updates": "Sprawdzanie aktualizacji",
    "Could not initialize HTTP library": "Nie można zainicjalizować biblioteki HTTP",
    "Downloading psp-updatelist.txt": "Pobieranie psp-updatelist.txt",
    "Got version %p @ %s": "Uzyskano wersję %p @ %s",
    "No need to update!": "Brak potrzeby aktualizacji!",
    "Downloading updater": "Pobieranie aktualizacji",
    "Running Updater": "Uruchamianie aktualizacji",

    "Skip gameboot": "Pomiń intro gry",
    "Language": "Język",
    "Font style": "Styl czcionki",
    "Scan savedata entries": "Skanuj zapisane dane",
    "Scan category entries": "Skanuj kategorie",
    "Scan DLC files": "Skanuj pliki DLC",
    "Swap X and O buttons": "Zamień przyciski X i O",
    "Menu animation": "Animacja menu",
    "Default Menu": "Domyślne menu",
    "Sort Entries by Name": "Sortuj według nazwy",
    "Show Recovery Menu": "Pokaż menu odzyskiwania",
    "Show FPS": "Pokaż FPS",
    "Text Glow": "Podświetlenie tekstu",
    "Screensaver time": "Czas wygaszacza ekranu",
    "Redirect ms0 to ef0": "Przekieruj ms0 do ef0",
    "Force Update": "Wymuś aktualizację",
    "Display Battery Percent": "Pokaż procent baterii",
    "Start Button Behavior": "Zachowanie przycisku Start",
    "System Menu Size": "Rozmiar menu systemowego",
    "Current": "Bieżący",
    "Last Game": "Ostatnia gra",
    "Random Game": "Losowa gra",
    "Small": "Mały",
    "Medium": "Średni",
    "Large": "Duży",

    "ARK Custom Firmware Settings": "Ustawienia ARK Custom Firmware",
    "CFW Settings": "Ustawienia CFW",
    "Launcher/Recovery Settings": "Ustawienia uruchamiania/odzyskiwania",
    "Menu Settings": "Ustawienia menu",
    "Installed Plugins": "Zainstalowane wtyczki",
    "Always": "Zawsze",
    "USB Charge": "Ładowanie przez USB",
    "PowerSave": "Oszczędzanie energii",
    "Autoboot Launcher": "Autouruchamianie",
    "Disable PSP Go Pause": "Wyłącz pauzę PSP Go",
    "Force Extra Memory": "Wymuś dodatkową pamięć",
    "Memory Stick Speedup": "Przyspieszenie Memory Stick",
    "Old Plugins on PSP Go": "Stare wtyczki na PSP Go",
    "Prevent hibernation deletion on PSP Go": "Zapobiegaj usuwaniu hibernacji na PSP Go",
    "Skip Sony logos in XMB": "Pomiń logo Sony w XMB",
    "Hide PIC0 and PIC1 in XMB": "Ukryj PIC0 i PIC1 w XMB",
    "Hide MAC Address": "Ukryj adres MAC",
    "Hide DLC": "Ukryj DLC",
    "Turn off LEDs": "Wyłącz diody LED",
    "VSH Region Change": "Zmiana regionu VSH",
    "UMD Region Change": "Zmiana regionu UMD"
}
`

krazynez, 2025-01-30:
Awesome thanks! Will update shortly.

krazynez, 2025-01-30:
In latest pre-release thank you for updating.

closed 2025-01-30 by krazynez