PSP-Archive/ARK-4 #474: [CL/Files] `Install Plugin` menu is missing `Launcher` option

Aneemasta, 2025-01-31:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

6.61 ARK 4.20.69 r154 cIPL (stable)

### Homebrew/GAME (if applicable)

Custom Launcher

### Plugins ( if applicable )

xmb, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, on
game, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, on
homebrew, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, off
homebrew, ms0:/SEPLUGINS/SSSSS/SSSSSBMP.PRX, on
ULJM05472, ms0:/SEPLUGINS/DIVAPATCH/DIVAPATCH.PRX, on
NPJH50465, ms0:/SEPLUGINS/DIVAXPATCH/DIVAPATCH.PRX, on
game, ms0:/SEPLUGINS/CUSHOME/CUSHOME.PRX, on
always, ms0:/SEPLUGINS/RJL/RemoteJoyLite.prx, on
launcher, ms0:/SEPLUGINS/RJL/RemoteJoyLite.prx, off

### Steps to reproduce

### Overview:
There used to be a separate `Launcher` option in the `*.PRX` cross-press menu. Moreover, its functionality is still there since you can manually type `launcher` as the first option in `PLUGINS.TXT` to have your `CL`-specific plugin manager task.
#### Prerequisites:
1. [SSSSS](https://www.gamebrew.org/wiki/Sssss_PSP) (JPG version) plugin unzipped as follows:
>`ms0:/SEPLUGINS/sssss/`
>` L  sssssjpg.prx`
>` L  sssssjpg.ini`
#### STR:
1. Run `CL`
2. Open `Files` menu
3. Navigate to `ms0:/SEPLUGINS/sssss`
4. Select `sssssjpg.prx` and press `cross`
#### What was expected:
- `Launcher` item present in the `Install Plugin` menu
#### What was happened:
- `Launcher` item absent in the `Install Plugin` menu

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

Aneemasta, 2025-01-31:
> launcher is just Custom Launcher. So it would just be `homebrew` or `game`
Yeah, I know it included in these filters. Still, you can add separate `launcher` filter within the older version, e.g. [20y. anniversary](https://github.com/PSP-Archive/ARK-4/releases/tag/r42069-151). It is useful to have the explicit ability to enable/disable plugins personally to CL since we already can enable/disable plugins per-game/brew with their IDs.

For instance, you can see my Plugins.txt with `RemoteJoyLite` enabled globally except for `CL` since it conflicts with USB storage mode.

krazynez, 2025-01-31:
> Yeah, I know it included in these filters. Still, you can add separate `launcher` filter within the older version, e.g. [20y. [...]

Yeah didn't know acid added that. My fault.

krazynez, 2025-02-03:
Added it back to Custom Launcher Plugin Installer. It is available in the latest release.

closed 2025-02-04 by JoseAaronLopezGarcia