PSP-Archive/ARK-4 #475: Read/Write VSH Settings to Internal Memory when no Memory Stick is inserted

wad11656, 2025-02-04:
### New Feature

On **ARK CFW**, when you make changes to the Advanced VSH Menu CFW settings...
[image]

...they get saved to `/SEPLUGINS/SETTINGS.txt`--a location on the Memory Stick:
[image]

On previous CFW's, changes in the **(Advanced) VSH Menu** saved to Internal Storage, so you weren't forced to keep the same Memory Stick inserted to retain the CFW changes you made in there.

Can we have ARK detect whether a Memory Stick is inserted, and if not, read/write the Advanced VSH menu settings there instead? At least the **VSH Region** and **UMD Region**?

When people ask me to unbrick their PSP's (without a Memory Stick), sometimes I like to adjust these settings before sending the PSP back. Also, personally, changes to the CFW and UMD regions seem like something that should stick whether or not the Memory Stick is inserted & working.

JoseAaronLopezGarcia, 2025-02-04:
There are a lot of risks that come with using internal flash memory for storing CFW settings:
- Potential for softbricks due to bad configurations.
- Potential for wasting the very little space available in flash.
- Potential for damaging flash due to constant rewrites to it.

People seem to forget that the PSP is a device from 2004, its internal flash memory is only 32MB and time will start (or has started) to take a toll on it sooner or later.

These are the reason why classic CFW like PRO and ME are riddled with softbricks, and why Recovery Menu was mandatory for them while optional for ARK.

That said, you CAN already do that with Full Flash Installation, where SETTINGS.TXT is installed to flash1:/ and from there you can change VSH and UMD region settings. I don't encourage to use flash1 settings due to the above mentioned potential problems. But I do allow more advanced users to make use of this feature if they wish so.
Though the UMD and VSH Region settings are in VSH Menu, they are actually CFW Settings (stored in SETTINGS.TXT instead of ARKMENU.BIN) so you can easily manually change them.

wad11656, 2025-02-04:
> I've made a small change to VSH Menu to let it use flash1 if it can't use regular settings.

Thanks!! I will need to test this out because I've been doing full-flash installation of ARK on multiple PSP's for several months, and the Advanced VSH Menu settings (with no Memory Stick inserted) never seem to stick. So maybe the `flash1:/SETTINGS.TXT` wasn't working for some reason?

JoseAaronLopezGarcia, 2025-02-04:
Give it a few minutes to compile and a new pre-release should be added. Let me know if it worked (I haven't tested myself, just wrote some code that *should* work).

JoseAaronLopezGarcia, 2025-02-04:
Did it work?

wad11656, 2025-02-04:
Sorry thank you I'll test within the next 18 hrs

JoseAaronLopezGarcia, 2025-02-09:
Marking this as complete unless something new comes up.

closed 2025-02-09 by JoseAaronLopezGarcia

hb-webdev, 2025-05-17:
Thank youuu!!! It's perfect!! I like it's in the CFW Settings. This change makes the CFW feel much more "solid" to me 😁

wad11656, 2025-05-17:
Thank youuu!!! It's perfect!! I like it's in the CFW Settings. This change makes the CFW feel much more "solid" to me 😁