PSP-Archive/ARK-4 #476: "ARK-4 Updater" in XMB Settings turns to "System Update" after updating ARK-4

Soto-Jnthan, 2025-02-04:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r155

### Steps to reproduce

1. Have access to a PSP 3000 with a fresh install of ARK-4 behind the most-recent release.
2. Confirm the ARK-4 Updater feature is available in the XMB settings.
3. Update to the most-recent release of ARK-4 using either the ARK-4 Updater feature or by copying the UPDATE folder to /PSP/GAME/ and running it.
4. After returning to the XMB once the update has finished, observe the change in name of the ARK-4 Updater feature to "System Update".

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

danssmnt, 2025-02-16:
Seems to be happening on PSVita (PCH-2016 model) too, I'm using the latest version (r163 as I write this)

JoseAaronLopezGarcia, 2025-03-16:
Could not reproduce in latest (pre)release.

Soto-Jnthan, 2025-03-22:
> Could not reproduce in latest (pre)release.

Hello, I can confim that this issue still occurs. I have updated via the internet from Rev 164 to Rev 167 and after the update was done, ARK-4 Updater turned into "System Update". It is worth mentioning that my PSP 3000 used ARK_cIPL and ARK_Full_Installer, as maybe the updating process handle those differently.

Soto-Jnthan, 2025-03-24:
What's the procedure when a bug has been mistakenly closed? Does it get reopened or shall a new issue be submitted referencing this one?

Soto-Jnthan, 2025-03-24:
Hello, these reproduction steps work 100% of the time. Let me know if there is something that is not clear:
 
1. Have access to a PSP 3000 with no CFW.
2. Download an ARK-4 version behind the latest release.
3. Move or copy ARK_01234 folder into /PSP/SAVEDATA/ folder.
4. Move or copy ARK_Loader, ARK_cIPL and ARK_Full_Installer folder into /PSP/GAME/ folder.5
5.  Run the following in sequential order: ARK_Loader > ARK_cIPL > ARK_Full_Installer.
6. Remove ARK_01234, ARK_Loader, ARK_cIPL and ARK_Full_Installer from the PSP.
7. Update to the latest release of ARK-4 via the internet using the ARK-4 Updater feature 
8. After returning to the XMB once the update has finished, observe the change in name of the ARK-4 Updater feature to "System Update".

You mentioned that the updating process is supposed to fail if the ARK_01234 folder is missing, but I thought that running the ARK_Full_Installer script should allow for the PSP to not have this folder anymore. Would it be too complicated for the updater to take into account that the PSP running the update has done the full installation?

JoseAaronLopezGarcia, 2025-03-24:
> Hello, these reproduction steps work 100% of the time. Let me know if there is something that is not clear:
> 
> 1. Have access to a PSP 3000 with no CFW.
> 2. Download an ARK-4 version behind the latest release.
> 3. Move or copy ARK_01234 folder into /PSP/SAVEDATA/ folder.
> 4. Move or copy ARK_Loader, ARK_cIPL and ARK_Full_Installer folder into /PSP/GAME/ folder.
> 5. Run the following in sequential order: ARK_Loader > ARK_cIPL > ARK_Full_Installer.
> 6. Remove ARK_01234, ARK_Loader, ARK_cIPL and ARK_Full_Installer from the PSP.
> 7. Update to the latest release of ARK-4 via the internet using the ARK-4 Updater feature
> 8. After returning to the XMB once the update has finished, observe the change in name of the ARK-4 Updater feature to "System Update".
> 
> You mentioned that the updating process is supposed to fail if the ARK_01234 folder is missing, but I thought that running the ARK_Full_Installer script should allow for the PSP to not have this folder anymore. Would it be too complicated for the updater to take into account that the PSP running the update has done the full installation?

I'll check it out.

krazynez, 2025-03-24:
Fixed in latest pre-release

closed 2025-03-24 by krazynez