PSP-Archive/ARK-4 #477: Minecraft PSP 3.4.2 - Doesn't show in Custom Launcher

omnismurfz, 2025-02-05:
### PSP/Vita Model

PSP-2000 (Slim), PSP-1000 (Fat)

### ARK Version

4.20.69

### Steps to reproduce

Putting Minecraft PSP 3.4.2 in the psp/game folder doesn't show in the custom launcher. Shows in the standard launcher no problem. 

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

the-POWERS, 2025-02-07:
same i installed the file in ppsspp but it dosen't show the icon/game

the-POWERS, 2025-02-07:
All you have to do: Unzip the game and that's it, if you're gonna play on psp put the files on a new Folder with the name of the game and move it to PSP/GAME There are two folders with those names on your psp.

omnismurfz, 2025-02-07:
I did do that. I can launch the game from the standard PSP menu, but it
won't show in the ark4 custom launcher menu.

the-POWERS, 2025-02-07:
> I did do that. I can launch the game from the standard PSP menu, but it
> won't show in the ark4 custom launcher menu.
> 
> 
> > All you have to do: Unzip the game and that's it, if you're gonna play on
> > psp put the files on a new Folder with the name of the game and move it to
> > PSP/GAME There are two folders with those names on your psp.
> >
> >
> idk abt that maybe try to join their own discord server 
https://discord.gg/W5rNjqW5

the-POWERS, 2025-02-07:
idk about that try to join their own discord server ig.
https://discord.gg/W5rNjqW5

Tammaxjnr, 2025-02-08:
I have the same issue with PS1 games not showing in Custom Launcher.

All CSO games are there and accessible/playable.
  
PS1 games are showing and playable through XMB/ Games/Memory Stick.

JoseAaronLopezGarcia, 2025-03-16:
Closing as can't reproduce in latest (pre)release.

closed 2025-03-16 by JoseAaronLopezGarcia