PSP-Archive/ARK-4 #478: Make the boot options split up

SerialKillerSlash, 2025-02-06:
### New Feature

What i mean is make the system boot animation have its own option for disabling and enabling and make it for gameboot too cuz i wanna make the system boot animation be the ony one enabled and the gameboot disabled, just like in other cfws

krazynez, 2025-02-06:
You can disable gameboot already

SerialKillerSlash, 2025-02-07:
wait really? i can still see skip sony logos as one option. if theres another way, let me know

JoseAaronLopezGarcia, 2025-02-07:
I will look into this.

JoseAaronLopezGarcia, 2025-02-09:
Done.

closed 2025-02-09 by JoseAaronLopezGarcia

SerialKillerSlash, 2025-02-10:
> Done.

Thanks.

SerialKillerSlash, 2025-02-11:
> Done.

uhh bro, u split the overclock option, i meant the boot animations, like in other cfws cuz they have skip sony startup and skip gameboot, thats what should be on ark-4, if ur reading this, take inspiration from other cfws, and this is one of them.

krazynez, 2025-02-11:
> > Done.
> 
> uhh bro, u split the overclock option, i meant the boot animations, like in other cfws cuz they have skip sony startup and skip gameboot, thats what should be on ark-4, if ur reading this, take inspiration from other cfws, and this is one of them.

Look in skip sony logos..

SerialKillerSlash, 2025-02-11:
> > > Done.
> > 
> > 
> > uhh bro, u split the overclock option, i meant the boot animations, like in other cfws cuz they have skip sony startup and skip gameboot, thats what should be on ark-4, if ur reading this, take inspiration from other cfws, and this is one of them.
> 
> Look in skip sony logos..

oh now i see, thx for the respond. shouldve seen instde it instead of the title, also how th did i get r161 early even though it hasnt been uploaded on the repository? strange

krazynez, 2025-02-11:
> > > > Done.
> > > 
> > > 
> > > uhh bro, u split the overclock option, i meant the boot animations, like in other cfws cuz they have skip sony startup and skip gameboot, thats what should be on ark-4, if ur reading this, take inspiration from other cfws, and this is one of them.
> > 
> > Look in skip sony logos..
> 
> oh now i see, thx for the respond. shouldve seen inside it instead of the title, also how th did i get r161 early even though it hasnt been uploaded on the repository? strange

it is on there