PSP-Archive/ARK-4 #482: New SSL root certificates to be added since ARK-4 r160 supports WPA2 now

dogaoz, 2025-02-10:
### New Feature

That way we can use PSP to access HTTPS(SSL) based websites in 2025.

JoseAaronLopezGarcia, 2025-02-10:
How will this help with anything? The PSP has 32MB of RAM, 64MB on later models, there's no way in hell you're gonna get most modern sites to work no matter how much hacking you do.

dogaoz, 2025-02-10:
If the following are fixed:

1. Root Certificates (SSL/TLS)
2. Native TLS 1.1, TLS 1.2, TLS 1.3 support

We can natively make HTTP GET and HTTP POST calls alongside enabling almost all websites in 2025 at least loading them up and enabling secure logins to websites etc. 

Of course, crazy complicated 40mb websites wouldn’t load properly due to RAM limitations.

However, we can also write homebrew apps to make API calls to ChatGPT, Spotify, or other latest/greatest software available today if TLS handshake worked for HTTPS based endpoints/sites.

This would enable limitless possibilities for all of us.

JoseAaronLopezGarcia, 2025-02-11:
We already have homebrew ports of TLS 1.3 and ChatGPT:
https://github.com/lorenzofelletti/chat_gpsp
I fail to see how any of this has to be "ARK's resposibility to provide a solution". We weren't even the ones who made WPA2 and we made that clear.

> We can natively make HTTP GET and HTTP POST calls alongside enabling almost all websites in 2025 at least loading them up and enabling secure logins to websites etc.

How the hell do you pretend to load all websites in 2025 using a device from 2004 that has RAM like a PC from 1995?

I don't wanna sound like that idiot from Blizzard, but don't you people have phones? Browse the internet there, or upgrade to a PS Vita if you really really really wanna browse the internet on an old portable device.

> If the following are fixed:
> 
> 1. Root Certificates (SSL/TLS)
> 2. Native TLS 1.1, TLS 1.2, TLS 1.3 support

No, you need a lot more than that to browse modern web pages, like a modern web browser, ideally one based on Chrome because Google forces a lot of websites to do shady anti-compatibility "bugs" on purpose to make them run worse on non-Chrome browsers.

JoseAaronLopezGarcia, 2025-03-16:
Closing as it's outside the scope of the project and WPA2 is experimental.

closed 2025-03-16 by JoseAaronLopezGarcia