PSP-Archive/ARK-4 #483: Squarez crashes on startup

SerialKillerSlash, 2025-02-11:
when i open up this homebrew called squarez, when i enter it the psp shuts down for some reason, why? i saw other ytbers playing it with no issue and i even tried it with the leda plugin.

JoseAaronLopezGarcia, 2025-02-12:
It works fine for me, make sure you download the latest version.

closed 2025-02-13 by JoseAaronLopezGarcia

SerialKillerSlash, 2025-02-15:
i downloaded the latest version and now it shows up as corrupted data despite the face i booted it up on the custom launcher via the files section and it worked great, why?

SerialKillerSlash, 2025-02-15:
> It works fine for me, make sure you download the latest version.

i downloaded the latest version and now it shows up as corrupted data despite the face i booted it up on the custom launcher via the files section and it worked great, why?