PSP-Archive/ARK-4 #484: Crashing when changing theme in Custom Launcher more than once

Creedoo, 2025-02-11:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r161

### Plugins ( if applicable )

game, ms0:/seplugins/RemoteJoyLite.prx, on
game, ms0:/seplugins/satellite_game.prx, on
game, ms0:/seplugins/MagicSave.prx, on
game, ms0:/seplugins/prxshot/prxshot.prx, on
vsh, ms0:/seplugins/RemoteJoyLite.prx, off
vsh, ms0:/seplugins/cxmb/cxmb.prx, on
vsh, ms0:/seplugins/categorieslite/category_lite.prx, on
vsh, ms0:/seplugins/dayviewer.prx, on
vsh, ms0:/seplugins/prxshot/prxshot.prx, on
pops, flash0:/kd/usb.prx, off
pops, ms0:/seplugins/RemoteJoyLite.prx, off

### Steps to reproduce

1. Navigate to Extras > Custom Launcher
2. In Custom Launcher navigate to your Themes folder
3. Install or preview a theme
4. Continue without installing the theme if previewing
5. Navigate to another theme folder
6. Install or preview the theme again
7. Observe the PSP freezing and then shutting down

I have reproduced the issue with CFW settings that would give the PSP a "performance boost":

- overclock for CPU clock in both XMB and games, 
- speeding up the MS access,
- and allowing the Extra Memory,

as well as without these settings. It seems that I have to completely reboot the PSP in order to have it not crash when applying more than two themes in one session, since exiting out of the Launcher and relaunching it does not prevent the issue from occurring. The theme also is not applied if the PSP crashes, forcing the user to redo the whole process. Previewing all of the themes this way is very time consuming and inconvenient.

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

Creedoo, 2025-02-12:
Sure, once I will be able to check it (in the next ~16 hours if that's okay, as I'm not at home right now unfortunately) I will let you know and close the issue if it'll be all good 😄

Creedoo, 2025-02-12:
Unfortunately the issue still occurs both when applying a theme when using one without installation, as well as installing second theme one after another in one session. I've updated ARK via internet after modifying the UPDATER.txt file to use the test server but I'm unsure if that update went through, as the version did not change 😅

JoseAaronLopezGarcia, 2025-03-16:
Working fine on latest (pre)release, I can change themes as I please many times.

closed 2025-03-16 by JoseAaronLopezGarcia

kondyan, 2025-06-21:
Hey, have the same issue on 180r version, try reseting CL settings first, then plug in your console, i plugged both data and charging cable and change theme. IDK how but it worked for me, hope you'll solve this problem