PSP-Archive/ARK-4 #485: Game list only shows partial games in XMB.

chenweiyj, 2025-02-12:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

Ark 4.20.69 r161

### Steps to reproduce

1. Open the game icon in XMB, and it only shows about 12 games and applications, not all my games in the /ISO folder. However, if I use custom launcher of Ark4, it can show and launch all my games.

So, how can I make the game list in XMB to show all my games in /ISO folder?

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-02-12:
How are these games installed?

chenweiyj, 2025-02-12:
> How are these games installed?

All games are iso format and downloaded by PC, and put into the /ISO folder. But only part of them are shown in XMB, whereas all of them can be accessed by Custom Launcher.

JoseAaronLopezGarcia, 2025-02-12:
Do you have other plugins installed?

chenweiyj, 2025-02-12:
> Do you have other plugins installed?

No

krazynez, 2025-02-12:
How many games are there. XMB has a limitation what I remember and why category lite would be the only way it displays all. Custom Launcher does not have said limit which is why it would work.

chenweiyj, 2025-02-12:
> How many games are there. XMB has a limitation what I remember and why category lite would be the only way it displays all. Custom Launcher does not have said limit which is why it would work.

about 17 games

chenweiyj, 2025-02-13:
> It's strange but I will try to reproduce it.

I have found the problem. If I put a video iso in the /ISO folder, it will affect showing all games. 

The following figure shows the unpack content of this video iso.
[image]

JoseAaronLopezGarcia, 2025-02-13:
Video ISO must go in the /ISO/VIDEO/ folder.

closed 2025-02-13 by JoseAaronLopezGarcia