PSP-Archive/ARK-4 #488: WiFi cannot reinitialise itself with WPA2 setting enabled

3zet, 2025-02-13:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.69 r162

### Steps to reproduce

It is same issue as with plugin but for completeness I will put it here. 

As in title, if you have an app like PKGi which connects to the internet straight away and you will left connection inactive for some time the console cannot reinitialise WiFi connection, you need to reconnect to WiFi from scratch again. My PSP has TA-079 board and at the moment I cannot test it on newer revisions. 

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-02-13:
We are not the devs of the WPA2 plugin/code and we do not provide user support as it's still in development.

closed 2025-02-13 by JoseAaronLopezGarcia

SteamRex25, 2025-02-15:
> We are not the devs of the WPA2 plugin/code and we do not provide user support as it's still in development.

who do I ask for support???