PSP-Archive/ARK-4 #492: [XMB/CFW Settings] Translation mismatch

Aneemasta, 2025-02-14:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

6.61 ARK 4.20.69 r162 [v1739533931](https://github.com/PSP-Archive/ARK-4/releases/tag/v1739533931)

### Homebrew/GAME (if applicable)

XMB

### Plugins ( if applicable )

xmb, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, on
game, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, on
homebrew, ms0:/SEPLUGINS/SSSSS/SSSSSPNG.PRX, off
homebrew, ms0:/SEPLUGINS/SSSSS/SSSSSBMP.PRX, on
ULJM05472, ms0:/SEPLUGINS/DIVAPATCH/DIVAPATCH.PRX, on
NPJH50465, ms0:/SEPLUGINS/DIVAXPATCH/DIVAPATCH.PRX, on
game, ms0:/SEPLUGINS/CUSHOME/CUSHOME.PRX, on
launcher, ms0:/SEPLUGINS/CUSHOME/CUSHOME.PRX, off

### Steps to reproduce

#### Overview:
Well guys, it seems that translations update effort #481 turned into a bit of mess. You can compare [XMB_EN.TXT](https://github.com/PSP-Archive/ARK-4/blob/main/extras/modules/xmbctrl/XMB_EN.TXT) to recently updated by me [XMB_RU.TXT](https://github.com/PSP-Archive/ARK-4/blob/main/extras/menus/arkMenu/themes/translations/resources/XMB_RU.TXT) to see the picture: the very amount and, therefore, order of strings aren't identical (for instance, `XMB_EN.TXT` has no `Prevent hibernation deletion on PSP Go` line).
This lead to some of `xmbctrl` translation lines being applied to wrong menu items.
#### STR:
1. Boot up PSP
2. In `XMB` open `Settings / System Settings / System Language` and select any lang other than English
3. In `XMB` open `Extra (or Game) / Custom Firmware Settings` and scroll down the entire menu to check items' translations
#### What was expected:
- Translations of menu items fit their places
#### What was happened:
- Menu items from (originally) `Use Extra Memory` to `Hide PIC0 and PIC1` translated with wrong text lines
P.S.: Your bug template seems to be quite outdated by now. Just saying 😅

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

Aneemasta, 2025-02-14:
#### Screenshots
1. English [image]
2. Spanish [image]

JoseAaronLopezGarcia, 2025-02-14:
You can find the proper language lines here: https://github.com/PSP-Archive/ARK-4/blob/main/extras/modules/xmbctrl/include/lang_en.h

Aneemasta, 2025-02-14:
> You can find the proper language lines here: https://github.com/PSP-Archive/ARK-4/blob/main/extras/modules/xmbctrl/include/lang_en.h

Oh, that`s handy. Seems like all of `xmb_<code>.txt` need lines reordering: placing current 10th line after current 13th.

Aneemasta, 2025-02-14:
@JoseAaronLopezGarcia, I think I'll be able to make a pull request for this task, but not right now (sleep time over there). You guys may wait for it if you prefer.

Aneemasta, 2025-02-15:
@JoseAaronLopezGarcia, I've checked latest release. There's more to fix, now lines 17 ("Prevent hibernation deletion") should be placed after line 14 ("Old Plugin Support on ef0"), and better double-check lines order afterwards.
Expect further pull request please.

JoseAaronLopezGarcia, 2025-02-15:
No problem, big thanks for the help.

Aneemasta, 2025-02-15:
Yes, this time it fixed.

closed 2025-02-15 by Aneemasta