PSP-Archive/ARK-4 #495: wpa2 wifi times out everytime on latest stabled

SteamRex25, 2025-02-15:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

rev 162

### Homebrew/GAME (if applicable)

N/A

### Plugins ( if applicable )

xmb, ms0:/SEPLUGINS/cxmb/cxmb.prx, on

always, ms0:/SEPLUGINS/battery_icon.prx, on

game, ms0:/SEPLUGINS/cushome/cushome.prx, on

### Steps to reproduce

1. setup network connection on psp
2. test connection
3. have it time out

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-02-15:
As stated: https://github.com/PSP-Archive/ARK-4/issues/488#issuecomment-2657770635

closed 2025-02-15 by krazynez

SteamRex25, 2025-02-15:
> As stated: [#488 (comment)](https://github.com/PSP-Archive/ARK-4/issues/488#issuecomment-2657770635)

well who do I get support from then?