PSP-Archive/ARK-4 #499: PSVita hangs when you exit the Recovery Menu (can't go back to VSH)

danssmnt, 2025-02-16:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

ARK 4.20.69 r163

### Plugins ( if applicable )

ULUS10041 ULUS10160, ms0:/seplugins/gta_remastered_v2.prx, on
ULUS10160, ms0:/seplugins/vcscheatdevice.prx, off
ULUS10041, ms0:/seplugins/lcscheatdevice.prx, off
ULUS10041 ULUS10160, ms0:/seplugins/cheatdevice_remastered.prx, off
ULUS10041 ULUS10160, ms0:/seplugins/cheatdevice_remastered_lite.prx, off
ULUS10114, ms0:/seplugins/CheatDeviceOwnTheCity.prx, off
xmb, ms0:/cxmb/cxmb.prx, off
xmb, ms0:/seplugins/category_lite.prx, on
xmb, ms0:/seplugins/dayviewer.prx, off
psp, ms0:/seplugins/atpro.prx, off
always, ms0:/seplugins/npdrm_free.prx, on
ULUS10285, ms0:/seplugins/cheatmenu_silenthill.prx, off

### Steps to reproduce

1. Boot ARK-4 from Bubble (created with [Adrenaline Bubble Manager](https://github.com/ONElua/AdrenalineBubbleManager/releases/tag/6.20))
2. Open VSH Menu inside XMB and go to Recovery Menu
3. Exit the Recovery Menu
4. PSVita hangs (doesn't go back to VSH, simply stays on a black screen)
5. By the way, ``Use Extra Memory`` is set to ``Auto``, ``Memory Stick Speedup`` is set to ``On``, ``Disable Analog Stick`` is set to ``Off`` and ``QA Flags`` is set to ``On``

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

danssmnt, 2025-02-16:
I've investigated a bit and found issue #465, which may be related, but isn't exactly the same

JoseAaronLopezGarcia, 2025-03-16:
Seems to be solved in latest (pre)release.

closed 2025-03-16 by JoseAaronLopezGarcia

danssmnt, 2025-03-22:
> Seems to be solved in latest (pre)release.

Can confirm it's solved!