PSP-Archive/ARK-4 #500: FTP server do not work with wpa2

Hetan-Layn, 2025-02-16:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK 4.20.69 r163

### Steps to reproduce

1.Activate WPA2
2.Go into ARK custom launcher
3.Start FTP server
4.Connect over WPA2 compatible network
5.Try to do stuff over FTP with an other device connected to the same network

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Always

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

Hetan-Layn, 2025-02-16:
Note: The same process work over none secure hotspot with wpa2 off

JoseAaronLopezGarcia, 2025-03-16:
Working fine here. In any case, WPA2 is known to be experimental and finicky, and so is FTP.

closed 2025-03-16 by JoseAaronLopezGarcia