PSP-Archive/ARK-4 #503: Power Stone Collection and Crazy Taxi - Fare Wars crash with error 80010016

JCP5009, 2025-02-18:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.69 r163

### Homebrew/GAME (if applicable)

Power Stone Collection (USA) [ULUS-10171] and Crazy Taxi - Fare Wars (USA, tested with versions 1.01 and 2.01) [ULUS-10273]

### Plugins ( if applicable )

game, ef0:/SEPLUGINS/NPLOADER.PRX, on
xmb, ef0:/SEPLUGINS/category_lite.prx, on
game, ef0:/SEPLUGINS/npdrm_free_mod.prx, on
xmb, ef0:/SEPLUGINS/npdrm_free_mod.prx, on
xmb, ef0:/SEPLUGINS/brightness.prx, on
game, ef0:/SEPLUGINS/brightness.prx, on
xmb, ef0:/SEPLUGINS/CXMB.PRX, on

### Steps to reproduce

1. Boot up PSP.
2. Open one of these 2 games, they should boot normally.
3. Select a individual game (Power Stone 1 / 2 or Crazy Taxi 1 / 2).
4. After some seconds, the system goes back to the XMB with the error "Game could not be started. (80010016)".

### USB Charge

Always

### Overclock

Disabled

### Power Save

XMB

### Balanced Energy Mode

Game

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JCP5009, 2025-02-18:
Nope, they are uncompressed ISOs and I checked that they are good dumps, I've turned off all the plugins and changed some settings and this still happens.

krazynez, 2025-02-19:
[image]

Working fine.

JCP5009, 2025-02-19:
Any configuration or plugin that you are using? I still get crashes consistently.

JoseAaronLopezGarcia, 2025-03-16:
This is most likely an issue of corrupted download or copy.

closed 2025-03-16 by JoseAaronLopezGarcia

D-113CR, 2025-04-29:
I got the same issue here, tested on multiple consoles including: psp go, psp 3000, and vita 2000 with no plugins loaded and with umd cache off

krazynez, 2025-04-29:
I once again tested both and both are working. No memory stick just on ef0 ( Internal Storage ).

JoseAaronLopezGarcia, 2025-04-29:
This was already fixed and it only affected PSP Go models.